bromagosa/Snap4Arduino #160: BT module - COM-port random disconnects

DenysZaytsev, 2017-05-27:
BT module HC-05/HC-06 disconnects randomly when script is running. Usually disconnects when no interaction from user, script is running and waiting for keys to be pressed. Then suddenly COM port disconnects. 
Tried HC-05/HC-06 modules and also SPP-CA module. Baud rate is set to 57600 on BT and on Arduino UNO board. No such issues when connected via USB cable. Also checked power. It is good, not related to this issue.

DenysZaytsev, 2017-05-27:
Tried to power up BT module with stand alone power source. Same issues. 
Script is actively running (switching to different blocks via broadcast command).
Tried connecting an Arduino without anything there (just arduino and BT). Tried Firmata, S5A and Firmata Plus) - same issues. 
Tried different BT modules. Nothing helps. Tried with Arduino Mega - same behavior

Is this an issue with BT modules (which are not supported) or Snap itself?

bromagosa, 2017-05-29:
Hi!

There must be something different on your side, we can't reproduce the problem here. Can you specify your OS, Snap4Arduino version, Arduino model (is it an original one?) and send us your Snap! project too?

Just for the record, I've had Snap4Arduino running over Bluetooth for a whole day uninterruptedly, so it's most probably not a Snap! issue :)

DenysZaytsev, 2017-05-29:
Hi bromagosa!

Thank you for quick reply. I'll let you know my configuration shortly.
Briefly it's laptop Acer Nitro with build-in bluetooth module and Windows 10 with all latest Microsoft's updates installed. I also tried to use USB Dlink DBT-122 adapter. 
Arduino is chineese clone from Aliexpress (Arduino Uno R3, also tried the same clone of Mega 2560). 
BT module I used was HC-05 (changed speed to 57600 and name, haven't touched any other settings ). 
I thought it might be happening due to a bad HC-05, but I tried a couple of them. 

Can we troubleshoot this somehow, as I need to make a lot of such sets for children learning?

I'll let you know more details shortly. I can either record a short video and post it somewhere.

My script is attached.

DenysZaytsev, 2017-05-29:
`<project name="UnoExperiments" app="Snap4Arduino 1.2.2&#xD; http://snap4arduino.org" version="1"><notes></notes><thumbnail>data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAKAAAAB4CAYAAAB1ovlvAAAgAElEQVR4Xu2dB3xV5d3Hv+ecu0dyMyEJI4EwBZkioFgn4gKsdVNHldbVOrA4arXiRmzdo1bbV9/WjUpxgKDsLUtWGBISSMhO7l7nnPfznBuiVn1NUXpLPM/nEy43ec7zf+7vfs//ef7POtKjjz66bNy4caMCgQA+n49DlRpXz6J81mNIUvssdOlaTMHV/0MymWTRokUMGjSInJyc9l18CHPF43HWrl3Lueeei6Ioh9DSj6NoqaysTF+3bh0DBgxAVVU+++wzamtr/yOfXtGTjAwson94wzfaq796EQsWLGDUqFHtrk902wKCM39PZ4+l3de0N2NmySAaznrUyC40GjNmTHsvNfN9iwLSxo0b9erqalwuFxkZGcbrwSar1crOmY8Q3LHyO4voPXIs1tGXGh5xe9l2xiy8Cix2bKfcSGLx8+jBBqqvnE99fT1er9cob9WqVYwYMeIrZYvfZWVl0b17d2w2G1Vv3c+J510BFhvRv1xs5HXd8CG4c5CdmSQ+fQPLkeNJbpyVet36EVJGJ/SGcuO9umcNeqQZdB2jcrpu/P1AWeU/f7/NvrhpzfT9FJDWrl2rf/7550Ypffv2RZIkdCH6QbxK6CQfPYmSLFu7arXnkg/a7Cx/63km9QbJmUFi2d+M6ysvm0M4HG4r65NPPqF3795EIhFEnW+44Qaeeuophg8fTs+ePQ2v1PDeI4wOLcE1dTHh6WNwXPQUcl5Pom/fhmRzYTvtViSbGz0eMl7VPZ8agEnuLOO91rgHpdtQQncPwjHpWfRoIAXgi5cY9Sg77+22+gwbNqxdn9PM9O0KSCtWrNCXLFnC8ccfb/RpBHwPPvgg2dnZjB8/ntmzZ+P3++nXrx+nnXYa06ZNY+zYsYhrRL9x0qRJRjM5efJkFEnC9+IZZDu/6Bu5pswnNusPqDsWf60WOy+c1Qb70rf/xqXRmV/J8/nFsw2o3G638fsrrrjC8IY33XQTt9xyC3PmzGHixIncfPPNnHzyyVRVVRH46AlO+e0zSBY7WlMl4UdOwnX9hyS3zSc+52Hs5/0x5fH8NcYrET+SrwCt7vPUezWO1lCB7CsAm7vt75GnJhp1+Gzia211/He6BiaE36yAtHjxYl38KRaL4fF4DI/0yCOPkJ+fz5QpU5gxYwYVFRVG0yeaOBGo7N+/n8bGRsPr1NTUcN555xlNd0tTE3kvT/wKgHJBPyRP7jcCuPmnb7R5wAWffMLU669uq2U8qbJk1Tqj2R09enS7vz/PlnfpU/5mu/P/OxmTnfqxYdjtxiXixhCBiJm+nwLS/PnzDQBFZHf00UcbpTkcDiMgEcCJqE+8iqZQeKJQKNTWTxTQir/Jsmz8Xkan2xsX4HO0Lzpcd+bf22rfMOdJMk78lfHe72/hjGOGoEkKSzfuMKLg4uJiioqKvt+n/QGuFnps3LiR3/72t1gsP3yg8wNU8bAqQmpsbNRnzZrFtm3bjGGOIUOGHPQHECCOKO2MQ9G+s4ygZmNNWYXRBM+dOxcRwIgk+lUTJkwwhzi+U8GOkUHSBQFmMhVIkwImgGkS3jSbUsAE0CQhrQqYAKZVftO4CaDJQFoVMAFMq/ymcRNAk4G0KmACmFb5TeMmgCYDaVXABDCt8pvGTQBNBtKqgAlgWuU3jZsAmgykVQETwLTKbxo3ATQZSKsCJoBpld80bgJoMpBWBUwA0yq/adwE0GQgrQqYAKZVftO4CaDJQFoVMAFMq/ymcRNAk4G0KmACmFb5TeMmgCYDaVXABDCt8pvGTQBNBtKqgAlgWuU3jZsAmgykVQETwLTKbxo3AWwXA+IAsXYe7/+t5f0QZbSrsodVJhPAdnxdOhroMpJ04CS7fx9GowzjNCj5+7PcjjofLllMANvzTek6uiQAUtBRAR1Zs4DcnouN7K3pS15QShrl/dhpNAH8DoZ0DSRZRUc8PUBGN34hGSjqkiBQTXk10UR/i2PUdNU4CztFoo6ktx7t++870nYSf/hkMwH8LgDF3w1uklTvK0PWVTp16Y8uK8iahC4gkjQkTRat9NcYFJeqOig6qbwkDRgFwmYyD6j8TgaMRlNNsrNsOYlIPTashOIuBh79EyTRLOtWVFlHPPUJLKmH23wpGScg6yrJaARVS+BwedCwGAe7m6mDA3jwIUMKDeG5hMcLteznnTdeZNDQ47ESZPPalZx+0Q1YPT6U1iO2dUlH/oZOoa6qBAP1VGxbhjhUvzkkMWLMeCQTwNagrMMeUi6CBvHspq96pPZ3u3RUNY6MhV2fzeXDuUu4dspdNPlrqd25msWLVnLlTdNEBxFJU9BFs/oNheuaxo7N83nt1Te46LzziTTtJav7cIpK+v/oA5DUqECHBbCt88aBcNVAUjwCDg1Nklr91bcjWVm2gl1bP2PJ0vnceOcTuF25Rn9vR9lKkpEadm3ex+mTrjH6dso3eD8NjWiwkSXv/Z1jxk7C5razdeWHBOtr6T/mHDKz81FJYJFFSKN875HGw7FJ75AAGt0uI3IQkWnKB4puv/idakSt4svWSCGYCiBQvgqiisaeTSsJ7t/I4tU7ueqW6SkPp2sIr7Z9zYfIxKmqi3PcmeeiiKDiX/t/SZ3ynct56uEHmf7nN9EVhVBzPXu3LqFyaxnHX/xrFKsHCypJUYKorlGN9vvpwxG6L9e5AwCYCi/FQG8KKIFZqvcnBo41TTx4MQWjgaBqQbKooMposoBTJomKbEArPFmq6dbUGI3Ve9i2cR4ZHheunIH0HDgsNSCtKySTAbatn8+Kjz5m8u0Pg243omBUDV1ubY61KCs/fgubsxODR56csi9D/f6dNOxezoZ1W8gq6Ifb6aawZwklpQNBsrYO2YjcHR/Fwx/A1khDIwmahUigmvfefJWTzvgZmhQ3HqAoHnodT4bRgnXGkzaj4TAeR4JAo0qmNxNLZh66pmL3OkkmJZx2B0uXLmbUmGNEY82GJXNwe7LoP+p07DlFOER0Iik0Nuxm4RvP8tNf3QeynBoXVFWDmpB/P/Pe+gs2tY6EJZvBo04mM6cnuppAV2RCNXuJBWp46c+Pc8Mf7ieuyeQU9sJizzLcoPDQKfw6tjc8zAFM0Sf+FYO9aBqv//0pHJLMKWeeacBkJUFSTaI3vYmqhUnGPOgRJ3FrEJvnCPR4PtithFWJvdVV/P2ll7ln2l00NDSR17nAaLKTwXrqa3azetEqfnHLDGSr1RjHCzTt452XH+Pnv35QzI0gImExMvPmiw/QNTNAc/Vm4rKbnoNP4dOl6xh/6a/RrW5sxEXPj9rdm/jzo3/CnlPM7XfcRkKSycguQlLsSOKG+hHMlBzmAAovIaPqGrLwGYkwDfX7jOZz/fLlHHXSWOrL5pPpycKdmcGKZ1+iaMRx7Hj3JUqHn8zeLZ8y8uqbkAsLScbiLFy8lPJd5Vxy2c955ZVXuPiiSagCbDXG9s2ryct2s2LVFi685nY0yUnDng289MQj3Dzjr6kHXKPw0mNTGdbLTYu/ispdGyntdyyO7FKsVpl/zprP1bdNR9MtxoMdWxrKeW76DMZd8At27NzBT885G02x483ojCSLPqVwhqYH/K/s54p+niTCT+P7SRh9uopd6wn5m1i3dh0/OWkc3mwvNO6kZvc8ZNtQ1vzlKcbf8ygb3nqT0Ka1qKEIqm4h55wL6H/yWGKBemSHnbKtOygtLSXoD5GVm2N40LgeZsuKOXjtbroPPIH8ol7s3rKQVQtXct51t6NjYfYrT9KryE+8rpb6yl1U19fR5+jjcHk6I1tdKJqNtRt2cdaF1yDLOk3N+3hi2r3cNv0x3nztDXr1KeXIwYNBtuPLLWjtyYpFEB0XwsPWA4q+maxKBII11Owrx6ZYWTTvLWIxlaZgAxMnnkNdbSP9Bw2nduNLeDv/hGxvAf98aBqDTz+D7fPex+fOpjkaY9Ck88gp7Itus/PHB+5h7Omn4fRkMPONt7jk8kvxen1oeox4oJb9e9azfOl6Lr32FpYtmc/QkRPIKSw0gpOHbjqfieedwP5dm6jYtQW3O4OufQbhtueiWV3IFoj5Azhz+9Kl12Aa927liT89xp3T/4Smydx39x+46qpfklPYGaxuMn25ICJ2E8D/NieooWoSsh5n22cL0BMSihZg84a1SIqT/ILuzP/ofSZfOwVXVh4QRk7GiFZ9hmIrQ7J3QgrWEItmYHP5kaQcglp3ZGs+2QUFLJo3j0HDj2bp4qUMGjSADZs2cMIJJ6DF4pRvX0U0kcDjzuCFv/0vj70wxwhAaqu38bcZd3D5dVOIBxqord9Oy9492J127Bk+XM5MVElBUjVWLVrPz66bSt32zXw4fymX/PIa4sLf6UmuvfIKnnjmz9hdLjSLiwxvDpIIbjqoEzxMPaAYixNfSpgNi97h84r9ZHslPt9eRm1dA4rdR36nzjhkna49ShkwfBRyMo4mW1FUP1piF4nIboLRrhCzIWVkofjrUO15JFQIhmNEIkGsiozL7cZq06lraCYUClJVXU5JUSEep0JzMpN+g0cSaqjjk9kvsGnFci66+maKevVBCzbSWFdB4/4qvFluHLId1elAV3WUWILHn3yBX1z5a+oDYfr074/V4caqOPB6M7j80st46slnaYm0kOHLpaC4tzHYLUkdbwHDYQmg0d8Xa/RI8uoz9zLu7HOZ+87LRAL7ye7UhaxOfenZuxdbP1vPhqWLKOk/AEmsv1MkuhX1o2rfHpJxlUA0DFoYh9WJxZuBx2YnmkwgK3Za6upxZPooKsynR0kpoUAYWVapq95JNBI0BqMLCvqSU9yPQF056z75H/bt3EHZjnrueOyvJCWZeKSJsL8Wf301LpcdyWJDiqhU1zcaS7xWrthB/yNHcsSR/YlqYtB6O4Vdi1izcQtqYz0XXnY+/nAzmbmlePNLUYzlXx0rHZYAaroIQFLN8KxXnqO4IIPFC+bSrcCBqmSBM49TTjuJT1esZ9Ens+nW40isWhRJkSjo2oOm2hrCoQDNkTBWyYXTrpCVmYk3NxurrKDYLFTt208i4keRbXQt6Yrf34BFibFj22aGDhmC05nLnPff51dT7iYYjVK+YTE7l79P4/697A04mfrwkyTVOLFwkETYT3PDPjas30Jx9xJ69Cimam85/3xzAYPGHEtu164cOWA4Sz5ZQH5RDi6vj/C+Ct569VWsdgfFfXrwy5sfwJaR2bHoO1znglMOUDdmHRJqhJnPTkOLVWGz2bA48/HmFpKdm8PKZavZvbWM4ccMJhmTUCxRYzC4vqEaRQuSiEpIThnF5qCTx4PN4UCWJRSnDS0aR5MU7E4nNq [... cut ...] cript><block s="doBroadcastAndWait"><l>Overload!</l></block></script></block></script></block-definition></blocks><variables><variable name="previousMillis"><l>0</l></variable><variable name="currentMillis"><l>1185.2</l></variable><variable name="ledState"><l>1</l></variable><variable name="interval"><l>0</l></variable><variable name="isLEDon"><bool>false</bool></variable><variable name="test1"><l>2.4755859375</l></variable><variable name="v"><l>7.4267578125</l></variable></variables></project>`

chrisclaynz, 2017-05-29:
hey mate - Are you using DC motors?  

Chris ClayIndependent Education Consultant@chrisclaynz  
Chris Clay | LEARNING TO THINK DIFFERENTLYChris Clay | LEARNING TO THINK
DIFFERENTLY While many are trying to encourage students to continue studying
STEM subjects it seems that the content of these courses remains unchanged. 
chrisclay.org

DenysZaytsev, 2017-06-06:
Hello there. I've managed to figure out exactly when disconnects happens.

So BT is being disconnected permanently only when there is an intensive data exchange between PC (host) and BT. When I'm turning ON/OFF PINs many times in the loop. As you can see from the screenshot - I'm just turning some PINs off in the EVER cycle. In this case LED on my BT module blinks all the time and after 10-30 seconds it disconnects from the host. 
[https://www.dropbox.com/s/s5usqygn73jc6xc/ScreenHunter_05%20Jun.%2006%2009.57.jpg?dl=0](url)

But I need to change the PIN state all the time. 
The project is RC Car and motor should only run when you press the key (i.e. up arrow) and HOLD it. When you release the key - motors should STOP.
Please note that disconnects happens EVEN when MOTORS ARE NOT running (when motors are stopped)!
Please see example of my script (see picture): https://www.dropbox.com/s/6h3ti9hwn7xbevv/ScreenHunter_07%20Jun.%2006%2010.02.jpg?dl=0

The logic is to set motor PINS to 0, when nothing is pressed in the ever cycle. But when arrow up is pressed - motors PINs sould be set to 1 in order to run the motors.

Yesterday I've even flashed RN-42 firmware to my HC-05 module, re-configured it from HID to SPP and tried again, but got the same behavior as before.

Please someone help me with it.

bromagosa, 2017-06-06:
Good that you figured this out, now it all makes sense! :)

You need to power the board and the motors using different sources. Otherwise the current drop when stopping the motor or the excessive current draw when starting it up will cause the BT module to disconnect. This has nothing to do with Snap4Arduino, it's all in the electronics :)

DenysZaytsev, 2017-06-06:
Hi bromagosa!

Can you share pls your IM (skype, viber, facebook...), I can share my screen for you, so you could take a look. 

The issue is not sticked to DC motors, as MOTORS are STOPPED when disconnect happens. It just disconnects when I'm setting PINs in the cycle.

DenysZaytsev, 2017-06-06:
I even tried to disconnect MOTORS physically from Arduino and still got disconnects when running the same script.

DenysZaytsev, 2017-06-06:
Guys, another question - is there a way to change communication speed between Arduino(I can change it in the Firmata sketch) and Snap4Arduino (at least Windows desktop application) ?
I tried to set baud rate in Firmata to 9600, but it was unable to connect after, obviously because desktop app (Snap4Arduino) is set to use 57600 baud rate.

jguille2, 2017-06-06:
Hi @DenysZaytsev,
As @bromagosa said, power source and electronics configuration is important, and this is causing the annoying disconnections (as you said, with USB connection all is right).
But also, you can reduce easily your board request!

You can use a 'wait' block to reduce these connections or, in the case you are commenting, you can avoid requesting when the state is not changing. Something like this:
[image]

DenysZaytsev, 2017-06-06:
> power source and electronics configuration is important
sure it is, but I tried to power up arduino and BT module from a power bank, when DC motors were powered with completely different power source (2 li-ion 18650 batteries) and behaviour was the same. 
Of cause I'll try to reduce connections, but it's weird that it works okay with the USB cord (Arduino and BT powered via USB and Serial link over USB cable as well), but doesn't work good with BT module (also powered separately from the motors).

How about changing the baud rate of Snap4Arduino for Windows ? How can I do it, I want to try

DenysZaytsev, 2017-06-06:
I think the problem is deeper - in the BT connection itself. I believe that 57600 baud rate is not good for Bluetooth to work without errors, so that's why I want to try 9600baud rate, but don't know how to configure the desktop application to work with this speed.

jguille2, 2017-06-06:
Or more elegant (and also natural):
[image]

And about changing baud rate... the only way is hacking Snap4Arduino. We are using Firmata protocol. You saw the config into the Arduino sketch and the server side is firmata.js int Snap4Arduino. I think you don't need to make this change.

DenysZaytsev, 2017-06-06:
Thank you, jguille2

That's exactly what I need. I'm gonna change the value in firmata.js and recompile Desktop app for Windows ... we'll seen then if that would make a trick.

Thanks a lot!

DenysZaytsev, 2017-07-11:
Eventually I was unable to use BT, so we decided to move onto the ESP8266 module. 
I've flashed it with esp-link firmware which does a transparent UART bridge, so it works good. No disconnect issues since.

However BT connectivity issues are still NOT solved.
Perhaps it depends on BT controller/drivers on the PC side, anyway it's still there.

closed 2023-05-25 by jguille2