bromagosa/Snap4Arduino #161: I have an issue about bluetooth connectivity with arduino uno

lsirikh, 2017-05-30:
Hi, I'm preparing to be a teacher for S/W education.

I am from South Korea. so, It would be awkward to writing skill. 

I have some problems.

First, I could not connect arduino (to which standardFirmata was uploaded) with bluetooth(bc-06) to snap4arduino.

As I already read some solutions about this, I couldn't fix it.

My bluetooth(BC-06) set by BAUD rate 57600.  My arduino was installed(uploaded) by standardFirmata.

 I used an external 9v square shaped battery(which is remain about 7v) for arduino power source. and I connect one servo to pin8 in arduino. 

Second, Sometimes I get connection with snap4arduino(intermittently). At that time I can control LED(PIN13) . but I couldn't control a servo at PIN8.

Is this problem from power source or other thing? 

I don't know what to do.

please help me.

jguille2, 2017-05-30:
Hi @lsirikh,

Yes, I think your problem is only with power supply. If you can connect (although it was sometimes), your config is right (bauds, firmware...)

The ideal would be an independent supply for the servo. But you can try to change the power source. A powerbank, or a stack of 1.5V batts will be better (I think 9V square batts is the worst option because its low maximun current)

Joan

lsirikh, 2017-05-30:
Thank you so much I will check the powerbank to replace the 9v battery. If I have other issue, I will remain more.

Thank you again.

closed 2017-05-30 by lsirikh