bromagosa/Snap4Arduino #162: How to resize  sprite size or the image I imported

lsirikh, 2017-05-30:
Hi, I'm preparing to be a teacher for physical computing education.

Snap4Arduino is a very good program to use for education. But I wanna use images and opensource hardware togegher. 

If I use only arduino, snap is pretty good. But I intended that students intercommunicate arduino(HW) with images on screen. to do this, I have to know the method to edit (;for example, resize, copy..) the images students imported. scratch or scratchx is pretty easy to edit images I imported and in Library.

I tried to use scratchx. but there are a lot of problems not to be solved, such as Bluetooth connection and browser dependency.

Does anybody know how to solve the issue, I'm really urgent.

Help me please~!

bromagosa, 2017-05-30:
To resize a sprite you can just use the [image] or [image] block.

Copying sprites is done by right-clicking on the sprite icon under the stage and selecting duplicate, unless you mean cloning them, which is done by using the [image] block.

I'm guessing you're new to Scratch and Snap!, since these are pretty well-known operations, so I'd suggest you take a look at the Snap! manual, where all of these are covered :)

Here's a link to it: http://snap.berkeley.edu/SnapManual.pdf

Cheers,
Bernat.

closed 2017-05-30 by bromagosa

lsirikh, 2017-05-30:
Thank you for answering my question.

What I mean is that I wanna know there are icons such as Duplicate, Delete, Grow, Shrink in the scratch nearby menu bar. that icon can edit image directly not by block coding.

Is it possible to do like this?

Anyway I really look up to you. It was an honor to get answer from you.

Thank you so much.

bromagosa, 2017-05-30:
Oh, I see what you mean now, sorry.

No, we don't have those in Snap!, but you can achieve the same results by using blocks and contextual menus :)

lsirikh, 2017-05-30:
I got it. 

Don't you have any plan to develop those kinds of functions in snap4arduino?

bromagosa, 2017-05-31:
I don't think so, I believe this is something that should be done in Snap! itself, but since there are other ways to achieve the same results these feel kind of redundant.