bromagosa/Snap4Arduino #175: Building on a Windows Machine

aramperez, 2017-07-23:
Is it possible to build Snap4Arduino unders Window 10 (Create)? The "Developers, read this" section of the Readme file appears to be written for a *nix system.

Thanks,
Aram

bromagosa, 2017-07-25:
Hi Aram, the build system is only working in GNU/Linux machines and (I think) MacOS. I wouldn't even know how to begin porting it to Windows, I haven't touched a Windows machine for a very long time :)

Why do you need to build Snap4Arduino yourself though?

aramperez, 2017-07-25:
Believe me, I prefer working on a Mac but for this project, I need to use Windows :-(

I would like to add some blocks for a robot I've built using a Maple Mini (ARM-based that can be programmed with the Arduino IDE).

I've installed Ubuntu on Windows and tried to build and I get the following error:
    =======
    Asking nwbuilder to build Snap4Arduino for win64
    =======

    src/platforms/desktop/win/64/build: 35: src/platforms/desktop/win/64/build:             
    environments/node_modules/nw-builder/bin/nwbuild: not found
    mv: cannot stat ‘releases/desktop/Snap4Arduino/win64/*’: No such file or directory
    desktop/win/64
    Removing temporary files...
    Copying license file over...
   Done!

I've started the process to build using VirtualBox, I'll let you know what happens.

Thanks!

bromagosa, 2017-07-25:
I think you may have not ran the prepare script before trying to build. Please check out this section in the README.md file:

https://github.com/bromagosa/Snap4Arduino#developers-read-this

By the way, you absolutely do not need to modify any source code to add blocks to Snap!, that's the whole point, you can Build Your Own Blocks ;)

aramperez, 2017-07-26:
Before posting, I followed all the instructions in your README.md file.

Respectfully, if you can build your own blocks, why did you develop Snap4Arduino? I'd like to talk to my robot over Bluetooth LE and NOT use Firmata.

aramperez, 2017-07-26:
Here's result of `./prepare --all` running on Ubuntu 15.04 under VirtualBox:

```ubuntu@ubuntu-VirtualBox:~/Snap4Arduino$ ./prepare --all
Pulling Snap! sources
Snap! sources up to date.
Pulling nwjs.io builder (nw-builder)
npm WARN engine nw-builder@3.4.1: wanted: {"node":">= 4.0.0"} (current: {"node":"0.10.25","npm":"1.4.21"})
npm ERR! git clone git@github.com:github:adam-lynch/node-temp Cloning into bare repository '/home/ubuntu/.npm/_git-remotes/git-github-com-github-adam-lynch-node-temp-f243469a'...
npm ERR! git clone git@github.com:github:adam-lynch/node-temp ssh: Could not resolve hostname github.com: Temporary failure in name resolution
npm ERR! git clone git@github.com:github:adam-lynch/node-temp fatal: Could not read from remote repository.
npm ERR! git clone git@github.com:github:adam-lynch/node-temp 
npm ERR! git clone git@github.com:github:adam-lynch/node-temp Please make sure you have the correct access rights
npm ERR! git clone git@github.com:github:adam-lynch/node-temp and the repository exists.
npm ERR! notarget No compatible version found: temp@'github:adam-lynch/node-temp#remove_tmpdir_dep'
npm ERR! notarget Valid install targets:
npm ERR! notarget ["0.2.0","0.3.0","0.4.0","0.5.0","0.5.1","0.6.0","0.7.0","0.8.0","0.8.1","0.8.2","0.8.3"]
npm ERR! notarget 
npm ERR! notarget This is most likely not a problem with npm itself.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.

npm ERR! System Linux 3.19.0-84-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "nw-builder"
npm ERR! cwd /home/ubuntu/Snap4Arduino/environments
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.4.21
npm ERR! code ETARGET
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/ubuntu/Snap4Arduino/environments/npm-debug.log
npm ERR! not ok code 0
ERROR: Could not install nw-builder. Won't be able to build any desktop version.
Please make sure that you have access to the Internet and both NodeJS and npm are installed in your system.
```

I get similar result using the new [Ubuntu on Windows](https://insights.ubuntu.com/2016/03/30/ubuntu-on-windows-the-ubuntu-userspace-for-windows-developers/).

bromagosa, 2017-07-26:
I developed Snap4Arduino because I needed more than just new blocks, otherwise I would have built a blocks library :)

You just said you needed a bunch of new blocks, that's why I suggested you didn't need to go through any build process, but it seems you actually need much more than new blocks.

Going through your error messages it seems that your Node version is older than the one `nw-builder` expects:

    npm WARN engine nw-builder@3.4.1: wanted: {"node":">= 4.0.0"} (current: {"node":"0.10.25","npm":"1.4.21"})

Try updating to a newer version of node (0.10.25 is pretty old) and see if that solves it.

Cheers,
Bernat.

aramperez, 2017-07-26:
I'm new to the Linux environment and using apt-get. I've tried to upgrade Node but haven't figured out how to do it yet. Thanks!

bromagosa, 2017-07-26:
I suggest you read through the NodeJS guide in https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions, and also find some kind of GNU/Linux introductory guide. Building projects from source is not something one usually does when first landing on a new operating system, so be ready for trouble! ;)

closed 2017-07-26 by bromagosa

aramperez, 2017-07-26:
I've done a bit of Linux programming (mainly with a Raspberry Pi), so I'm not a complete novice.

So I found how to upgrade my nodejs and ./profile --all worked. I built the desktop/win/64 and a Snap4Arduino.exe appears in the build 'releases/desktop/win/64' folder. When I try to run it, I get an "Invalid argument" error. Any thoughts?

P.S. I need receive an error about some sort of key of Chrome but I'm ignoring that for now.

bromagosa, 2017-07-27:
Great, that's good progress! Can you share a screenshot of the error?

aramperez, 2017-07-27:
Here's the warning about Chrome:
[image]

Here's the error from the building the desktop version (`./build --platform=desktop/win/64`):

[image]

bromagosa, 2017-07-28:
You're trying to run Snap4Arduino, Windows version, from within GNU/Linux.

You need to run that file on your Windows machine :)

sxcem, 2019-09-25:
https://github.com/bromagosa/Snap4Arduino/issues/291

I  build  success ,write a buglist :)