bromagosa/Snap4Arduino #177: https://snap4arduino.org needed

ToonTalk, 2017-08-04:
An HTTPS connection is necessary to access devices such as the microphone or camera. Also for communicating with various web services. I get an error from https://snap4arduino.org/run

Because we are relying upon some blocks defined in JavaScript and need to run on the Raspberry Pi we need to rely upon the Chrome extension.

https://snap.berkeley.edu/snapsource/snap.html works fine.

jguille2, 2017-10-27:
Hi @ToonTalk,

This is fixed with the domain change (.org -> .rocks)

You can open [http://snap4arduino.rocks/run/](http://snap4arduino.rocks/run/) or [https://snap4arduino.rocks/run/](https://snap4arduino.rocks/run/)

closed 2017-10-27 by jguille2

ToonTalk, 2017-10-28:
Glad to hear but Chrome 62  says:

This server could not prove that it is snap4arduino.rocks; its security certificate is from *.github.com. This may be caused by a misconfiguration or an attacker intercepting your connection.

FireFox 56 says:

snap4arduino.rocks uses an invalid security certificate. The certificate is only valid for the following names: *.github.com, github.com, *.github.io, github.io Error code: SSL_ERROR_BAD_CERT_DOMAIN

jguille2, 2017-10-28:
Yes @ToonTalk,
We'll considere the need to have ssl certificates... but I guess you already know you can ignore this warning and use it 'normally'.
In chromium:
[image]
and firefox:
[image]