bromagosa/Snap4Arduino #178: Chrome extension fails to connect to Arduino while Windows 64-bit version works fine

ToonTalk, 2017-08-04:
I get "Could not talk to Arduino in port COM5"

ToonTalk, 2017-08-08:
This problem persists despite the fact that when I went to chrome://extensions the Snap4Arduino connector wasn't enabled. Enabling it caused the 'Connect Arduino' button to offer a choice of ports. (Today it is COM3.) But the same behaviour as before and yet the desktop executable version connects fine.

Could Chrome have changed since on my previous laptop both the web and desktop version worked fine? And previously when I followed the instructions at http://snap4arduino.org/#plugin I didn't need to explicitly enable the extension. 

Using Chrome Version 60.0.3112.90 (Official Build) (64-bit)

ToonTalk, 2017-09-09:
Today I reproduced the inability to connect using the plugin on a Raspberry Pi running the latest Raspbian. The Chrome app did connect OK.

jguille2, 2017-09-11:
Hi @ToonTalk,
We were working into the web_chromium version... and there will be changes in the next release.
This working is not finished, but *firmata* is already updated and connectivity is better.

If you want to test this, you can download the plugin [here](https://github.com/jguille2/SA5/files/1291015/Snap4Arduino.crx_dev.zip), and use the dev online version in [http://snap4arduino.rocks/run_dev/](http://snap4arduino.rocks/run_dev/)

Joan

ToonTalk, 2017-09-11:
Thanks. I got the plugin to work but not reliably. And even after closing the tab, the Arduino IDE failed with an error message about the port being busy.  Perhaps the plugin connected OK but didn't close the connection?

Using the plugin I tried to connect to an Arduino, succeeded, disconnected (via the button), tried to reconnect and it failed.

jguille2, 2017-09-13:
I'm sorry... I can only test with UNO boards. I will try to get some other model to do more tests.

bromagosa, 2017-09-15:
It may be that newer Chrome versions don't let us use extensions in the background, I've had a similar experience when building another extension.

I'll take a look and let you know, thanks for reporting!

closed 2023-05-25 by jguille2