bromagosa/Snap4Arduino #180: Problems with http server

ToonTalk, 2017-08-15:
After enabling the http server (which by the way should be mentioned on http://snap4arduino.org/#http) I see

[image]

Entering these URLs into any browser (I tried Chrome, FireFox, and Edge) always takes me to the search page. When I tried to open one of these URLs from inside of Snap (using the http block) I see an error "".

[image]

However when I tried http://localhost:42001/broadcast=m it worked

ToonTalk, 2017-08-15:
Also is it possible for the Snap4Arduino Chrome extension to support the http server?

jguille2, 2018-02-13:
Hey,
I close this because the issue is fixed in 1.2.5 (next Friday!) and now that dialog shows only ipv4 IPs.

And I had already written down the possibility of adding httpserver feature into web version (into the plugin)... That is in my TODO large list :)

closed 2018-02-13 by jguille2