bromagosa/Snap4Arduino #181: Chrome extension should respond when unable to connect to Arduino

ToonTalk, 2017-08-15:
If a visitor to http://snap4arduino.org/run/ is using a browser other than Chrome or is using Chrome but hasn't installed and enabled the extension then clicking on 'Connect Arduino' should inform the user that they need to use Chrome and to install and enabled the extension. Currently it does nothing.

ToonTalk, 2017-08-21:
This problem continues to persist for a MKR1000.

With a Mega 2560 I had a somewhat similar problem (the connecting message stayed up for many minutes before giving up). But the Arduino IDE gave me an error message that indicated there was an issue with file privileges. Running the IDE as administrator (Windows 10) fixed the problem and fixed the problem with the Chrome extension -- it is working for this Arduino but not the MKR1000 (which is did before with a different Windows laptop). The desktop Snap4Arduino app connects to the MKR1000 but then after the first command says it is disconnected.

So maybe this issue is that the MKR1000 is acting poorly. In which case the only minor complaint about Snap4Arduino is that it isn't giving useful error messages when trying to connect to an Arduino.

jguille2, 2017-10-29:
Hi @ToonTalk,

Current chromium version is quite different from the status of your reported issue. There is a newer Firmata version... and many connections problems have dissapeared.

I close this issue. Feel free to reopen it (or open another one) to report new connections problems.

Thanks,

Joan

closed 2017-10-29 by jguille2

ToonTalk, 2017-10-29:
Thanks. It is working much more reliably now.