bromagosa/Snap4Arduino #182: Embedded Command Line Version problems with http requests

julienrat, 2017-09-01:
Hi !
I have made a little program who send http requests every seconds like this : https://img11.hostingpics.net/pics/349364tlchargement.png

When i  run CLI on xml script : `./snap.js snap.xml --plain-snap`
I have this error : 
---------------------------
snap/threads.js:2140
        this.httpRequest = new XMLHttpRequest();
                               ^

ReferenceError: XMLHttpRequest is not defined
    at Process.reportURL (snap/threads.js:2140:32)
    at Process.evaluateBlock (snap/threads.js:675:32)
    at Process.evaluateContext (snap/threads.js:636:21)
    at Process.runStep (snap/threads.js:553:14)
    at snap/threads.js:300:18
    at Array.forEach (<anonymous>)
    at ThreadManager.step (snap/threads.js:298:20)
    at StageMorph.step (snap/objects.js:5730:22)
    at StageMorph.Morph.stepFrame (snap/morphic.js:2893:14)
    at snap/morphic.js:2895:19

bromagosa, 2017-09-04:
Thanks for reporting this, I'm taking a look right now :)

bromagosa, 2017-09-04:
Please get the patched version from http://snap4arduino.org/downloads/1.2.3/Snap4Arduino_embedded-cli_1.2.3.zip

I'll update the GitHub release in a minute too.

closed 2017-09-04 by bromagosa

julienrat, 2017-09-04:
Hi it works only one time,
i have got this script 
https://img11.hostingpics.net/pics/831367tlchargement.png

and when i run it `./snap.js jardin.xml --plain-snap`

it works on time and i have got this error

Lutin says: Led2 allumée
Lutin says: Led2 allumée
threads.js:17
                    myself.httpRequest.response = str;
                                                ^

TypeError: Cannot set property 'response' of null
    at IncomingMessage.<anonymous> (threads.js:17:49)
    at emitNone (events.js:110:20)
    at IncomingMessage.emit (events.js:207:7)
    at endReadableNT (_stream_readable.js:1059:12)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)

bromagosa, 2017-09-04:
You're totally right, that was a careless slip on my side.

Please try it again :)

julienrat, 2017-09-04:
Excellent !
Thx again for your reactivity ! It works !

commit f04b960c (2017-09-04): fixes #182, plus reduces CLI package size by a factor of 10