bromagosa/Snap4Arduino #184: Network Serial doesn't work in new version of Snap4Arduino

DenysZaytsev, 2017-09-08:
Hi Guys,

I used Snap4Arduino with Network Serial Ports enabled on version 1.2.2 and worked like a charm. 
Recently I've installed the latest Snap4Arduino version 1.2.3 on my laptop and Network Connection doesn't work there.

Please see screenshots for refference. 
[https://www.dropbox.com/s/zxxqeu8oaiejkmo/SPA_4.jpg?dl=1](url)
Also in latest version 1.2.3 - if you enter Network address manually it still connects to the default pre-defined address instead - tcp://arduino.local:23 no matter what you enter in the feild:
[https://www.dropbox.com/s/d65jbq164k4ivs9/SPA_2.jpg?dl=1](url)
[https://www.dropbox.com/s/akuc4yi9283izhn/SPA_3.jpg?dl=1](url)

overisberg, 2017-12-09:
Network Serial Ports works in version 1.2.4 but it is not possible to change the hostname.

In an earlier version I changed it to tcp://bjorn.local:23 and even if I try to change it to tcp://arduino.local:23 it tries to connect to tcp://bjorn.local:23.
On another computer where I have not used Network Serial Ports earlier the hostname tcp://arduino.local:23.

I think the problematic change was in this commit:
https://github.com/bromagosa/Snap4Arduino/commit/a3eb3453c8131875124c604ad2189af8fbbdad37#diff-cbc86da9e2bca459215166dd01100580

jguille2, 2017-12-11:
Hi, @overisberg, you are right.
Next version will fix this problem.

Hi, @DenysZaytsev, and excuse our delay. I am not using Network Connection and then, I can not test it... but if your problem was only with the hostname management, it will be fixed soon.

closed 2017-12-12 by bromagosa

ZCB1208, 2017-12-22:
@overisberg I'm also trying to make a connection between esp-link and snap4arduino .
I try to use the code you submit and compile it again.
But it still doesn't work.Don't know what's going on.
The version I compiled just can't run.I don't even know where the problem is.
The computer I use is macbook.
It will be very appreciat if you can share more about esp-link and snap4arduino
I am a student from China.  My English is poor .  
I'm sorry, the above is translated by computer.   
it may make you a little confused.

jguille2, 2017-12-22:
Ups!
I'm sorry but we have problems to test Macs versions (we haven't got any). We know that we have some issues with the nwbuilder in the MacVersions, and in the releases @bromagosa does some extra actions to fix this (with the help of some mac users).

Maybe Bernat could explain some details of these extra actions and you could test it later.

ZCB1208, 2017-12-22:
This is the first time I have given a question at github, and the first time I received a reply.Thank you very much for your reply to my question.  There is really no need to say sorry .The work you do is really great ,Thank you very much indeed.  I'll try my best  to make it work. @jguille2

PR #212 (2017-12-12): Fixing #184 networkConnection issue
It fixes #184 problem.
Currently, network connection is always using the same _hostName:port_ (the default _arduino.local:23_ or other 'custom default' used in previous versions (before 1.2.3)) and we can not change it.