bromagosa/Snap4Arduino #186: Servo affecting DC motor operation

chrisclaynz, 2017-09-13:
'm noticing a consistent problem occurring when using a servo motor in addition to DC motors.

It's weird and my friends and I have all racked our brains but come to nothing.

We have a little robot with 2 DC motors, an ultrasound module and a servo running. A script has been written that has the bot drive forwards till it's Ultrasound mod detects an object. At this point it stops and the servo rotates right and left to identify the best way to turn. 

The issue we have is that once the servo works the right motor stops getting any power. The only way to remedy the problem is to increase the power the motor receives to 255 (254 does nothing...). 

Once this problem has occurred it will remain even if it's disconnected (via bluetooth) and reconnected (even using a different script). The only way the right motor is once again able to work with less than 255 is to reboot the arduino...

Any help with this would be most appreciated.

jguille2, 2017-09-13:
Hi @chrisclaynz,
I think your problem is with the board limitations. Arduino uses *timers* to do different tasks (basically oscillations like in servos, pwm, tone...). Some functions use the same _timer_ and therefore can not be performed simultaneously.

For example, UNO uses _Timer1_ with Servo library. Also _Timer1_ is needed to do PWM in pins 9 and 10. Then, if you use servos, you can not use PWM with pins 9 and 10.

Maybe you can change the pins of the pwm connection of your motor driver.

Joan

chrisclaynz, 2017-09-13:
Thanks for this Joan. I'll give it a try.

Does this mean that it's not possible to run dc motors and servos
simultaneously while using snap4arduino?

Cheers,

Chris

jguille2, 2017-09-13:
No! We can use servos and dc motors simultaneously.

What I've said is about board limitations (not from Snap4Arduino... also with Arduino sketches). My hypothesis (I don't know your connections) is that this was the problem.

Then, if you use servos , you can not use pwm in pins 9 and 10 (you can use both pins as standard digital pins). But if you need pwm operation to control motor speed, you can use other pwm pins (in UNO 3, 5, 6 and 11).
Note that if you use another board, timers and pins are others.
If you want to share your elctronic configuration, we can add more specific details.

Joan

chrisclaynz, 2017-09-18:
Ahh...
So I'm using a motor shield with an UNO that is hardwired to pins 10, 11, 12 and 13 :(
I'm wondering how I might work around this...?
Ideas in my head include:
1. Always run the motors at full speed when using a servo
2. Use a separate motor driver module instead of the shield - messy wires :(  

Any alternatives would be most welcome... What setup are you using Joan?

jguille2, 2017-09-18:
Hi @chrisclaynz,

I need more info about your motor shield...

If you this shield controls two DC motors (usually, two motors are controled by 6 pins) then I guess pins 10 and 11 are controling motor speed. If so, you can control the speed of 1 motor (the one that uses the pin 11) and the other motor only works with full speed.

Report us more details about your shield, I we can think of more possibilities...

chrisclaynz, 2017-09-18:
Thanks for getting back to me Joan.

The current setup has 10 and 11 as PWM pins for each of the two motors
whilst 12 and 13 are the digital pins which control the direction in which
the motors move (forward or reverse).

From what you say I could use just pin 11 for the PWM so both motors can
move at the same speed whilst direction can be controlled with 12 and 13.

Is this correct..? This sounds like a fair compromise but I'm still
interested to know more about your own setup.

Thanks again mate!

aramperez, 2017-09-18:
IMO, unless your motors are of very high quality, controlling both with just one PWM will not spin them at the same speed.

jguille2, 2017-09-18:
Yes, @aramperez is right.
But if you are able to change hardware configuration (change the connections from Arduino pins to the motor driver), you can reconnect the pwm (original in 10) to other pwm pin (3, 5 or 6). Then you will control one motor with 11 pin (normal mode) and the other with the pin you have chosen

chrisclaynz, 2017-09-18:
Thanks again @adamperez and Joan.

Joan- when you say change the hardware configuration I'm assuming this will
mean changing to a different motor shield right?

jguille2, 2017-09-18:
Hi,
I've said 'if you are able to change hardware configuration' because you said before 'I could use just pin 11 for the PWM so both motors can move at the same speed'. And for this, sure, you must change the connections!

In any case, two more comments...
  - When you buy another shield, yes, you can take care of this, and make sure that the pwm of the motor control speed are not 9 and 10.
  - But maybe you can fix this easily by changing the connections. If you don't want to manipulate the shield (soldering...) you can easily act on the connection between board and shield.
    For example:
[image]
    In this picture, the pin "10" of the shield is not connected to the Arduino  pin 10. It is connected to the pin 3.
    It's only an example. In this, you are losing Arduino pin 10... but you can easily rescue it with a wire... And of course you must be sure the shield is not using its pin 3 for other features... If so, you can also bend pin 3 connection and do it directly with the wire (from board to shield).

That's all!

chrisclaynz, 2017-09-18:
Thanks again Joan- really helpful!

I'll give this a try. I am yet to find a motor shield that doesn't use pins
10-13. Please let me know if you know if one.

Thanks again!

jguille2, 2017-09-19:
OK I think we can close this issue.

Just insist again...  is "a motor shield that does not use pins 9 and 10 as pwm pins" (for control speed). It can use them for digital control (direction, stop...)

I think most of the shields are like this. My first searches on Google shows me:
  - [Arduino Motor Shield rev3](https://store.arduino.cc/usa/arduino-motor-shield-rev3) uses pins 3 and 11 for pwm-speed.
  - [DFRobot Motor Shield](https://www.dfrobot.com/product-69.html) uses pins 5 and 6.
  - [Velleman Motor Shield](http://www.velleman.co.uk/contents/en-uk/p580_ka03.html) where pwm pins can be selected with [jumpers](http://www.velleman.co.uk/manuals/ka03d.pdf) on the shield.
  And sure, you can find cheaper shields...

That's all,

Joan

closed 2017-09-19 by jguille2

chrisclaynz, 2017-09-19:
Thanks again Joan!