bromagosa/Snap4Arduino #187: How to translate snap4arduino to other language?

JavohirYakubov, 2017-09-17:


jguille2, 2017-09-17:
Hi @JavohirYakubov ,

What is your language?

  - If this language is not translated into Snap!, you must do it into the [Snap! project](https://github.com/jmoenig/Snap--Build-Your-Own-Blocks). We can help you and report more details about this.

  - I this is already translated in Snap! and you are asking about the translation of the specific elements of Snap4Arduino, then, here is the right place. You can use [catalan translation](https://github.com/bromagosa/Snap4Arduino/blob/master/src/s4a/lang-ca.js) as the template. You will see English strings... and change catalan translations to your own language.
You can make a Pull Request into this git repository. If you are not a git user, you can send us your file an we will implement the tranlation to the repo.

Ask us any questions...

Joan

jguille2, 2017-10-29:
Hi @JavohirYakubov,

I close this issue. Any question will be wellcome.

Joan

closed 2017-10-29 by jguille2