bromagosa/Snap4Arduino #188: Malfunctioning Snap4Arduino on the tablet Androin.

YuriZel, 2017-09-19:
Good afternoon. I apologize for my English (google translator). When working with blocks, for example, movement, other blocks are not dragged, for example control, sensors and so on! What will you advise? Thank you in advance!

bromagosa, 2017-09-19:
Hi!

The Android version is very buggy and experimental, sorry about that. We're actually thinking of discontinuing it and building a project loader instead, so that you can write your Snap4Arduino programs using a regular computer and then export them to the tablet, but at the moment we don't have much time to spend on this :(

YuriZel, 2017-09-19:
It is a pity that you want to complete the project, I'm very comfortable using the tablet at work, but you can not use a computer! Sorry for concern!