bromagosa/Snap4Arduino #189: Ignoring Arduino Custom Block Code During Arduino Code Export

iceinstructor, 2017-09-20:
The code export for Arduino works fairly well except when using custom blocks. It seems that the custom blocks are not being checked during the export of the *.ino file.

jguille2, 2017-09-20:
Yes, you're right... but this will not be fixed.

Into [Snap4Arduino features](http://snap4arduino.rocks/#feature), we have _Transpilation of simple scripts into Arduino sketches_ but we can not transpile every Snap4Arduino project. And we can not do this because we can not transpile a dynamic and liveness language into an Arduino sketch. We can not transpile parallelism, recursion, abstraction (our custom blocks), OO, first class objects... And we can not use transpilation with all Snap powerful: other scripts, other sprites, other blocks (that they are hidden ... but they are still available because Snap! does not close doors), libraries...

Many times we have been about to remove this functionality because it does not respond to Snap's paradigm (a live environment), but many people use it to obtain stand-alone sketches ... but it can only do this: transpile simple scritps.

Joan

closed 2017-09-20 by jguille2

iceinstructor, 2017-09-21:
I hope you don't remove the Arduino export feature as it is one of the key benefits of the software so students don't need wireless or a tether. These features help to set it aside from the other similar tools and help to reduce cost of hardware/robotics platforms. As an Engineer promoting STEM I am very glad this tool was made. Students and parents alike are able to use it with minimal introduction. I sure hope your team considers expanding the Arduino export feature to include the custom blocks. The advantage from an educational perspective is the ability to introduce programming at various levels of abstraction.

Thank you.

jguille2, 2017-09-22:
Hi @iceinstructor ,

Don't worry. Export feature works well enough so it can help in some cases of use ... even if it is not the Snap! operation paradigm... and I think it will be maintained into Snap4Arduino project.

At Snap! we all are unconditional enthusiasts of Abstraction. Then I fully understand your demand. But we are working into a dynamic and liveness environment (we are also unconditional enthusiast of this). And therefore, a dynamic environment can not be translated univocally into a static one. At no time does a pre-evaluation of the code (this allows us dynamism, recursion, parallelism ...).

It might be worthwhile to give Snap paradigm an opportunity. It's just an opinion ... but I would like to point out some comments:
  - We do Computer Science projects. Yes, we need a PC. We will not always need a board... some times we will only need a virtual-digital environment, others we will want to interact with our environment (with the board or with the PC peripherals (keyboard, mouse, webcam ...)), and I will also have Internet to interact with the world wide.
  - With Snap4Arduino we can easily connect Arduino boards... but we can use other devices: Makey-makey, Picaxe boards (with S2P software), Legos devices (with S2Bot soft)...
  - We don't have a single program (like a sketch). We have projects with many elements and many scripts: different operations, testing, debuging... And we can run/stop them and change them at run time.
  - And if the project needs to connect wirelessly... we have different options. The main one is do the connection by bluetooth.

Ok... I just wanted to share some ideas. Let's continue!!

Joan