bromagosa/Snap4Arduino #192: Block combinations causing bluetooth issues

chrisclaynz, 2017-09-24:
Hey all. 
Today I had created a simple rc style robot with x2 dc motors controlled with keystrokes (simple if/then blocks within a forever loop).
To ensure the motors stopped when no key was pressed we used the "not" block from the operator section and added a "any key pressed" block from the sensing section. 
When we have this stop script the bluetooth connection is real bad and the robot quickly disconnects. However, when this is removed bluetooth is absolutely fine. 
I think this problem may exist with other combinations but I'm baffled as to why this would be the case. 

Any ideas most appreciated. 

Chris

jguille2, 2017-09-24:
Hi Chris!
Is not useful for you the comments of [this other issue](https://github.com/bromagosa/Snap4Arduino/issues/160) ?

As we said, power source and electronics configuration is important and you can do changes to improve your operation... but I think is better to try to minimize board requests. It does not make sense to be communicating continuously to the board (motors) that does not have to do anything!

Many code options... You can use [_repeat until_ blocks](https://github.com/bromagosa/Snap4Arduino/issues/160#issuecomment-306516594) or [_wait until blocks_](https://github.com/bromagosa/Snap4Arduino/issues/160#issuecomment-306530104). Then, the communications to the board will be momentary and the rest of the time the code will be into the _wait block_

I hope you find this useful....

Joan

adryannn, 2018-12-01:
Hi chrisclaynz ,

I had the same issue with my robot. Initially I was using a power bank (5v)  which could not provide enough power, so when motors started the bluetooth loosed the connectivity. After added 7.4v (2x3.7v ) or even 6v (4x1.5V), the BT connectivity remains  stable.
By the way I'm using either the arduino uno VIN power input  or arduino uno shield which require higher voltage due to the regulator.

Hope that works for you.

Kind regards,
Adrian