bromagosa/Snap4Arduino #196: deb package for armhf

openelearning, 2017-10-16:
Hello,

I'm trying to deploy an open source, open hardware and low cost solution for schools using H3 single boards, that can give the opportunity to children to make DIY activites (cf. https://forum.armbian.com/index.php?/topic/4929-which-card-choose-to-deploy-an-educational-open-source-os-in-schools/&tab=comments#comment-37500 ). If Snap4arduino could run on it, without internet connexion, that could be great :) I'm try to use armbian ubuntu xenial desktop legacy, that needs armhf deb package, using Orange Pi PC+ SBC.

The chromium solution of snap4arduino seems to need internet connexion. 
The command line solution of snap4arduino seems to not provide an UI.

So it seems that the best way in order that the children can use Snap4arduino on OrangePiPC+tu upload code on arduino boards  without internet connexion seems to have a deb or appimage package that is armhf friendly.

Do you think it's to possible to build one (or because of some bugs or other reasons, it's impossible to make armhf package of Snap4arduino) ? 
Is it easy to learn how to build it ? (I'm newbie, I tried to make this https://build.opensuse.org/package/show/home:Ucino:snap4arduino/snap4arduino but it doesn't work :) ) 
What is the easiest way to have Snap4arduino running without internet connexion on a H3 SBC ?

Thanks a lot !
Best regards

bromagosa, 2017-10-17:
Hi!

Unfortunately, there is no officially supported way to build Snap4Arduino for armhf architectures. The issue comes from nwjs.io not supporting it. There have been some unofficial efforts to do so, but they keep getting discontinued as new versions of nwjs.io come out :(

openelearning, 2017-10-17:
Hi!
Ok thanks a lot for your reply. Hmm... it seems that arm SBC is not actually a good choice to make these computers, too much softwares doesn't work on it.

closed 2023-05-25 by jguille2