bromagosa/Snap4Arduino #201: Transpiring Snap4Arduino to an Arduino Sketch expects Snap! variable names to follow C syntax

ToonTalk, 2017-10-28:
E.g. a variable cannot be called "pin number" with a space.

jguille2, 2017-10-28:
Hi @ToonTalk,
First, a comment. We aware that we can not translate a dynamic and liveness language into an static sketch. In this way, we'll try to do our best to make simple sketches, but there will be always pending situations. You know custom blocks (one of Snap! main features) are not transpilable.
For this issue:
  - I think the better solutions to this issue is replace white spaces (' ') with '_' character. Then if I use "pin value" and "pin_value" there will be a conflict. Maybe we can also replace '_' with '__' to avoid this problem.
  - Also, related with #194, I'll change 'i' var (index into the 'repeat' block) with a less common name (maybe 'loopIndex') to avoid collisions with a hipothetical global variable named 'i'.

Continue...

ToonTalk, 2017-10-29:
The subset of the language that appears when one opens a translatable project sure looks like it should be completely translatable. While you are right that in general local custom blocks can't be translated I wonder if there aren't cases where it could be. Maybe codification support could be used in those cases. (I see that the codification support option is checked but it doesn't seem to have the usual blocks and menu items.)

Replacing white spaces in variable names sounds like a good idea.

Note that #194 isn't just caused by 'i' but multiple 'repeats' leads to multiple declarations. I assume this would happen with 'loopindex' as well.

jguille2, 2017-10-29:
Hi @ToonTalk,
  - Yes, #194 isn't caused by the 'i' name, and that issue will be fixed by #202 (avoiding global declaration). But fixing that and contemplating the global subject of variables, I think changing 'i' name will avoid potential collisions with an 'i' named variable used into the _repeat_ block. Then I'll choose 'loopIndex' or a more "stranger" name...

  - We can not close Snap! features to make a new IDE translatable to Processing. You can do it with Blockly or another tool (Ardublocks, Visualino...)... and you'll be written Processing with a graphical tool.

    We have custom blocks, and we can always use recursion. And you can make parallel jobs everywhere. And many of our objects are first class...and our variables are not typed! And everyone can visualize the hidden primitives and start doing totally untranslated things...

    Many times we have considered removing this tools (transpiler) to avoid these problems, but we keep it because it is a useful tool. But we are programming Snap! not Processing.

Thanks,

Joan

ToonTalk, 2017-10-29:
Is recursion a problem? I just tested a sketch with recursion and it worked fine. The Codification in the Examples area translates factorial and doubly recursive Fibonacci to C just fine. (True someone needs to add code generators for those functions.)

And the parallel jobs aren't a problem since the interface to translation is to produce a single sketch from a single 'When green flag clicked' block.

I'm a bit confused about your references to Processing. Arduino sketches are in a C/C++ type language.

ToonTalk, 2017-10-29:
Here's a little test of recursive user defined blocks. It generates a working Arduino sketch:

[image]

[recursive flicker project.zip](https://github.com/bromagosa/Snap4Arduino/files/1425145/recursive.flicker.project.zip)

closed 2023-05-25 by jguille2