bromagosa/Snap4Arduino #204: Improve timer accuracy

seteahconnoc, 2017-11-01:
I am using Snap4Arduino to measure the dark time of a forked light barrier. Obviously, the timer continuously counts upwards by tenths of seconds. Is there a method to achieve higher accuracy? In the source code of "Snap!" itself, I was able to get the appropriate result by changing the function "getTimer" in "objects.js". Sadly, I do not succeed in reaching this with "Snap4Arduino". Any suggestions would be great.

jguille2, 2017-11-01:
Hi @seteahconnoc ,
Two different problems here: 1- Snap! timer accuracy and 2- Latency with board communication
### 1 Snap! timer accuracy
Snap! timer accuracy  has an easy solution. You said you hacked getTimer function in objects.js. Snap4Arduino is an Snap! Then, the solution is the same... but certainly you must build the app (it is a node app... and you can't change directly js sources)

But Snap! is great! And you don't need to hack anything! Because you have JS functions available into Snap! blocks!!! For example, you can do:
[image]
or you can delete the Math.floor rounding...
### 2 Board latency
But if you need a high timer accuracy and you are reading board sensors... maybe the latency with the board communication will not allow this wanted resolution. If this is the case, you can use [pulseIn Arduino function](https://www.arduino.cc/en/Reference/PulseIn) with SA5 firmata (then the time is measured by the board). And if your measurement is more complex, the only way will be another firmata modification.

Ok, I think this is all. You can close both issues... And if you have problems with pulseIn, we can continue in https://github.com/jguille2/SA5/issues/4.

Joan

seteahconnoc, 2017-11-02:
This sound quite plausible. I guess, it 's just the small technicalities that are getting in the way. So, I am very thankful for your help.

closed 2017-11-02 by seteahconnoc