bromagosa/Snap4Arduino #205: Help me please!!  Snap4Arduino  can't via Bluetooth

caimunnoi, 2017-11-05:
[image]

Board Arduino LEONARDO R3
HC-06 JY-MCU
Battery 9V

MacOS Sierra 10.12.6
Snap4Arduino  1.2.4
Arduino IDE 1.8.3

StandardFirmata
Baud rate 57600
NAME Bluetooth  WHATEVER

I follow this step by : https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Blueetooth

caimunnoi, 2017-11-07:
Thank you bromagosa for answer

jguille2, 2017-11-09:
Hi @caimunnoi ,
Then, is working for you when you are plugged? And can you test this with other boards?

I'm sorry but I haven't got any Leonardo board... and I can't test your configuration.

Tell us your results... maybe we can write down more details... or we can close this issue.

Thanks,

Joan

jpliew, 2018-02-05:
I can confirm that Leonardo is working properly using USB cable. @caimunnoi  you might want to check with USB cable first like @jguille2 said.