bromagosa/Snap4Arduino #206: Long start

raffoux, 2017-11-08:
Hello

Since snaparduino version 1.2.3, the software takes a long time to start, it is probably caused by the change of the address of the snaparduino website. The software tries to download a file at the old address.

[image]

I use the 32 or 64-bit linux version with my students.
Thank you for this excellent software.

jguille2, 2017-11-08:
Yes, you are right.

But you must upgrade to [1.2.4](https://github.com/bromagosa/Snap4Arduino/releases/tag/1.2.4) (released in October 19th) that fixes this delay caused by the website domain change.

Thanks,
Joan

closed 2017-11-08 by jguille2

raffoux, 2017-11-16:
Thanks,
Jean-Christophe