bromagosa/Snap4Arduino #209: CodeMaster – Automatic Assessment and Grading of Snap! and App Inventor Programs

JeanHauck, 2017-11-30:
Hello,

we are excited to launch and share our CodeMaster, a free web application that in a problem-based learning context allows to automatically assess and grade projects programmed with App Inventor and Snap!.

CodeMaster is available in English and Brazilian Portuguese.

More information: http://www.computacaonaescola.ufsc.br/?page_id=2523&lang=en

Or just try it out: http://apps.computacaonaescola.ufsc.br:8080

The tool has been developed by two students as part of their final undergraduate thesis in the context of our initiative Computação na Escola at the Federal University of Santa Catarina/Brazil. Based on first feedback, we are planning to continue the development of the tool as part of future student’s theses. Any suggestions are very welcome.

Best regards,

Christiane, Aldo and Jean

jguille2, 2017-11-30:
Hi Jean,
Thanks for your report.
We saw your notice and we are testing (you know Snap4Arduino team are all Snapers!!)

Surely we can use your work... and we will send our feedback.
Go on and thanks for your work! We are convinced that Snap! is a wonderful tool for education.

Joan

closed 2017-11-30 by jguille2

JeanHauck, 2017-11-30:
Thanks @jguille2 !