bromagosa/Snap4Arduino #210: Does not run on MacOS 10.13.1

neuwirthe, 2017-12-01:
I am trying to run Snap4Arduino (downloaded the binary from snap4arduino.rocks) om MacOS 10.13.1,
but I cannot be started. I get an error:
The application "Snap4Arduino.app" can't be started -47
How can the issue be resolved?

neuwirthe, 2017-12-02:
It turns out that the error occurred because there was another instance of Snap4Arduino already running. So it was my fault, but the error message could possibly be more informative.

bromagosa, 2017-12-04:
Hi Erich!

I'm afraid this error message comes from nwjs.io, the underlying engine that runs Snap4Arduino, and I don't think it can be changed. I will take a look anyway.

Thanks for reporting! :)

closed 2023-05-25 by jguille2