bromagosa/Snap4Arduino #211: Can't connect to Arduino in Debian

jljvdm, 2017-12-10:
Connecting to Arduino doesn't succeed on a Debian Linux. Linux  4.2.0-1-amd64.
Is uploaded the firmware en I tested it. It runs perfectly

Test beginPrintsVersion passed.
Test defaultDigitalWritePortIsZero passed.
Test processWriteDigital_0 passed.
Test processWriteDigital_127 passed.
Test processWriteDigital_128 passed.
Test processWriteLargestDigitalValue passed.
Test receivedStringShouldDecodeFrom2BytesPerChar passed.
Test sendStringShouldEncode2BytesPerChar passed.
Test setFirmwareVersionDoesNotLeakMemory passed.
Test specifiedDigitalWritePort passed.
Test summary: 10 passed, 0 failed, and 0 skipped, out of 10 test(s).

When i start Snap4Arduino and i connect to the arduino then is can't connect. 
The led on the board is lightning once and then the error message shows "Could not talk to Arduino in port /dev/ttyACM0 Check if firmata is loaded". After that the port is dissapeard in Snap4Arduino.

Can somebody help me?

jguille2, 2017-12-11:
Hi Jeroen (@jljvdm),

I'm a debian user and I'm connecting to arduino with Snap4Arduino without problems.

Can you report more details?  I'm going with some notes:
  - Snap4Arduino version? I guess 1.2.4 Desktop-64bits
  - Arduino board? I suggest testing with an Arduino UNO to leave board issues aside.
  - Board firmware? We need the StandardFirmata.ino form an official IDE release.
  - Is your debian user into the _dialout_ group? It needs permission to access to serial port device.

Let's continue...
Joan

jljvdm, 2017-12-11:
Hi Joan ( @jguille2 ),

Thanks for your help.

My Snap4Arduino is version 1.2.4
I used two different Arduino's. The Nano and the Meag 256R3. Both of them didn't function.
The latest Firmata is used as i downloaded with the Arduino IDE. I used the unittests for testing and the 
My user is in the dialout group.
jeroen : jeroen adm tty uucp dialout cdrom floppy audio dip video plugdev users netdev lpadmin bluetooth scanner docker

Do i need some libraries?

jljvdm, 2017-12-11:
I tried a firmatatester. https://github.com/firmata/firmata_test/downloads
It worked like a charm. The led13 set low annd high and the led on the board blnked.

jguille2, 2017-12-11:
Hi Jeroen @jljvdm,

No, you don't need extra libs. Everything seems to be right.  I do not have a good hypothesis ... but I try to comment several things to see if we can unblock this situation.
  - Maybe your board is busy with another request? Make sure you have closed other programs (or browser plugins) that connect with the board.
  - Anyway, try unplugging and plugging again before the Snap4Arduino connection.
  - It is also good to do a Reset (Board button). You can also reset immediately after the Snap4Arduino  connection. Try different resets.
  - As you have seen, when you receive the "Check if firmata is loaded" message, you can't reconnect the board and you have to close and reopen Snap4Arduino. This is a known issue in Snap4Arduino.

Of course, all these comments are not the normal behavior... but sometimes they help us to solve strange situations.

Two extra comments:
  - I insist in test with an official Arduino IDE. I say this because I have seen strange behaviors with modified IDEs ( for commercial reasons, to run ok only with their boards).
  - The Snap4Arduino Desktop version is the reference (and the best option). But you can test with Snap4Arduino online and the _chromium_ plugin to test and see what results you get.

Continue...
Joan

jljvdm, 2017-12-11:
When I uploaded the firmata there is a led lights up on the board. 
When I start Snap4Arduino and I try to connect to the Arduino board the led light dims.

jguille2, 2017-12-11:
This confuses me...

You upload StandardFirmata.ino, right?

Which LED is ON when you plug the board... and then it dims when connected?

With StandardFirmata you should have all the LEDs OFF at the beginning (except the "ON" led, of course).
During the connection with Snap4Arduino "RX" and "TX" Leds will blink... but all the others will remain OFF.

closed 2023-05-25 by jguille2