bromagosa/Snap4Arduino #214: Can't connect to Arduino on Linux

fachat, 2017-12-20:
I know this might be a duplicate to #211 but with an own issue we can track and report progress if it isn't.

I am running Linux kubuntu 16.04.3 LTS. I just downloaded the Arduino IDE, and Snap4Arduino from scratch, so they are current versions (Arduino 1.8.5, Snap4Arduino 1.2.4).

I installed the firmata firmware through the Arduino IDE multiple times now. Everything looks good. I even tested it with a firmata test tool and was able to successfully control some pins I tested on the Arduino nano I have connected to /dev/ttyUSB0. 

(In the beginning the yellow Arduino LED is off, the red and green LEDs are on) 
When I start Snap4Arduino with the launcher.sh, I can select the "Arduino" section of the palette, and click on "Connect Arduino". Then the yellow LED starts flickering (while the green gets dimmer), then the yellow one goes off and the green gets bright again. A moment later I get the message "Could not talk to Arduino in port /dev/ttyUSB0. Check if firmata is loaded". After that, clicking on "Connect Arduino" has no effect whatsoever. I need to leave Snap4Arduino and restart it to try it again.

Frankly, I am running out of ideas what to test, esp. as the firmata tester was able to switch LEDs on/off immediately.

fachat, 2017-12-20:
I tried downgrading to Snap4Arduino 1.2.3, but even after removing ~/.cache/Snap4Arduino and ~/.config/Snap4Arduino it only shows and empty window? Removing cache and config also does not help fix the problem with 1.2.4

Also note there is a Bluetooth module on the board itself, but it is not connected to any host (mobile phone). It has a jumper to set either "AT" or "COM" mode. Both settings result in the same error situation as described (even with the jumper header removed the error stays).

fachat, 2017-12-20:
Is there any way to get some debugging log output or similar?

bromagosa, 2017-12-21:
Which board are you using? Is it a Leonardo by any chance?

jguille2, 2017-12-21:
Hi @fachat,

We need to know witch is your board, and witch are these leds (in your board, or maybe in a shield over your board).

If the green LED is the "ON" pin, it is always lightning when there are power supply. Then, if this led is getting dimmer... the problem is with the power supply. And if this board has a BT module, you must add more power supply to have an stable serial connection.

Try to add an extra power supply (not only the USB connection).

Joan

fachat, 2017-12-21:
I think this is a board compatible with the Arduino Nano. It is from the Conrad IoT Advent calendar ( https://www.conrad.de/de/adventskalender-conrad-components-internet-of-things-experimente-bausaetze-ab-14-jahre-1540862.html ), and looks like this board (it has the name Franzis printed on it: https://www.elv.de/franzis-arduino-internet-of-things-bluething-board.html )

The yellow LED is connected to pin D13, the red and green LEDs seem to be connected to D11 and D12 (can't right now tell in what order without starting it up again).

jguille2, 2017-12-22:
Ok,
So, led green is not the ON led. Green al red ard the RX and TX fins of the BT.

Sorry, I have not a quick solution but I share some comments:
  - Maybe this board is not an ideal for working with Snap4Arduino. You can't use your integrated BT. We usually use BT modules, but only connected to D0 and D1, to replace the USB connection.
  - The power supply could be a problem... but you have only a power supply connector (the usb) it is busy with the serial connection to the PC.

  - Can you test to do a reset in the middle of the connection? Sometimes this helps to get the connection, avoiding interferences (BT).

bpatricksullivan, 2018-08-07:
I am having this same issue. I am using an Arduino UNO with 64 bit Snap4Arduino 4.1.2.7 on 64 bit Ubuntu 16.04. I uploaded StandardFirmata to this Arduino successfully both with this (linux) computer, and with a Windows computer, both using Arduino IDE 1.8.5. On the Windows computer Snap4Arduino connects and runs Snap4Arduino programs with no issues, but I receive the "Could not talk to Arduino in port /dev/ttyACM0 Check if firmata is loaded." on the Linux machine every time. I am in the dialout group. I have tried doing a reset in the middle of the connection and have tried plugging in the Arduino before and after starting up Snap4Arduino.

jguille2, 2018-08-09:
Hi @bpatricksullivan,

Snap4Arduino developers are Linux users... so Linux is the most tested OS.

I can't reproduce this issue... so we need more testing...

- Launch Snap4Arduino from the console.
  - With the console into the Snap4Arduino folder, you can type `./run`
  - Do your test and try the connection as ususal
  - If the error continues ('Could not talk to Arduio in....'), then look at the console for more info about the problem.

- Second test. I know your in the dialout group and permissions must be ok, but do the same test as a superuser: With the console in the Snap4Arduino folder, type `sudo ./run`, enter your password and try the same test.

Report your results and we'll try to see the problem...

Joan

bpatricksullivan, 2018-08-10:
Thank you so much @jguille2 !
After running Snap4Arduino from the command line, I am now successfully connecting to the board. It was not necessary to run as superuser. 

I'm now generally able to connect to the board without running from the command line, too. I'm still experiencing some intermittent  'Could not talk to Arduino in....' message if I close and reopen Snap4Arduino, but unplugging the board, connecting once from Arduino IDE, and then reconnecting from Snap4Arduino seems to resolve the problem. I wonder if the port is not being freed if I close the window without clicking Disconnect Arduino?

Moltes gràcies for the help and for developing this tool. I'm teaching an Arduino based class to 13 year olds this fall. They mostly have Chromebooks and have programmed in Scratch in the past, so this is a perfect transitional tool. 

salutacions cordials,
Brian

closed 2023-05-25 by jguille2