bromagosa/Snap4Arduino #216: Connected predicate?

DavidGasku, 2018-01-11:
Sometimes I create a lesson where the student can simulate some program with some virtual hardware inside snap4arduino, and it would work even without the arduino board.

In order to do so, I create a flag variable "connected". When it's off, it will activate the virtual led, for example, if it's on, both virtual and real. This way they can program something that I built, and they test it when I plug it in their computers.

It would be very handy to have a predicate "arduino connected?" built in, so I can use it in my custom blocks.

I don't know how would it work with several instances of arduino, but I would like to start such discussion.

Thanks!

jguille2, 2018-01-11:
Hi @DavidGasku,
Good idea!

Many projects have a virtual representation of the physical system. I also often use a variable to indicate if I want to work virtually (without having the board connected) or not.

But it makes a lot of sense to detect simply if there is a connected board. Then...

- You can build this block
[image]

- There is no problem with several instances. The block looks at the connection of its sprite.

- Here you have the xml with this block. Maybe we can think to add it as a primitive (@bromagosa ?)
[arduinoConnected.xml.zip](https://github.com/bromagosa/Snap4Arduino/files/1624457/arduinoConnected.xml.zip)
- Then, when we work with simulations, we can do something like this:
[image]

Joan

DavidGasku, 2018-01-12:
Great! Thank you very much, it works :)

bromagosa, 2018-01-15:
You don't need to escalate the morphic hierarchy, the context of a JS block is always the receiving sprite, so you can just do:

[image]

We could add this as a primitive, why not? I'll do that :)

jguille2, 2018-01-15:
Thanks Bernat,
I think this can be useful. I've see your [implementation](4fc5a3e8f2713d48dacdab1acb8f882e344b0b23), so I can close this issue.

closed 2018-01-15 by jguille2