bromagosa/Snap4Arduino #217: Plugin not working on https://snap4arduino.rocks/run/

ToonTalk, 2018-01-12:
While it works on http://snap4arduino.rocks/run/

We need https for various reasons.

The error is the same as #128  "Error in event handler for (unknown): TypeError: Cannot read property 'forEach' of undefined"

I see that the website it says "If you prefer to run it in your own server, use this downloadable version." -- The link is to a zip file with the CRX file as well as www folder with the whole system. Is the CRX file one that doesn't have the whitelist problem mentioned in #128?

jguille2, 2018-01-12:
Hi @ToonTalk 

I see this problem.

We'll fix this for the next release. And I will post here a crx modification (today or tomorrow) to fix it in the meantime.

Joan

jguille2, 2018-01-13:
Hi @ToonTalk ,

This issue with the operation of the chromium plugin over _https_ on snap4arduino.rocks is already fixed for the next version.
In the meantime, you can update the chromium plugin with [this 1.2.4+ version](http://snap4arduino.rocks/files/Snap4Arduino.crx). This is already published on [snap4arduino.rocks/#plugin](http://snap4arduino.rocks/#plugin) web.

So, I close this issue. Thanks for reporting it!

Joan

closed 2018-01-13 by jguille2

ToonTalk, 2018-01-13:
Great, thanks. Working fine now.