bromagosa/Snap4Arduino #219: "arduino not connected" using "Tell Sprite to"

DavidGasku, 2018-01-12:
I have one object connected to arduino. If another object wants to interact with it, I use the "Tell to" with some arduino block. I would expect this block to be run under the arduino sprite's context, but it is throwing me the "arduino not connected" error.

[image]

Is this supposed to work like this? If so, any way to workaround this problem?

Thanks!

jguille2, 2018-01-12:
Thanks @DavidGasku for your report.

I see the problem!

We must to change our check
```javascript
var sprite = this.homeContext.receiver;
```
for this other
```javascript
var sprite = this.blockReceiver();
```
and then, calling an Arduino function from other Sprite will run ok.

I'll change this and it will be fix in the next version (soon).

Joan

DavidGasku, 2018-01-12:
Thanks Joan.

jguille2, 2018-01-13:
The issue is fixed for the next version. It will be soon to be able to enjoy with Snap 4.1.x

So, I close this issue. Thanks for reporting it!

Joan

closed 2018-01-13 by jguille2