bromagosa/Snap4Arduino #221: Pressure sensor BMP180

seteahconnoc, 2018-01-13:
Hello.

I am trying to use the pressure sensor BMP180 with Snap4Arduino.

My internet search lead me to this thread
https://groups.google.com/forum/#!msg/s4a-community/sNhqLSwHd74/nSiPFDOKDQAJ
in the "S4A Community Mail List". There, the issue remained unsolved. "jguille2" wrote: "... the problem is you have to open Sparkfun library and program all the actions and calculations into Sn4A. I have not much time and also I haven't this sensor to test it... but if you want to work with it, we can help you..."

Any further suggestions would be welcome.

Thanks.

bromagosa, 2018-01-19:
Hi @seteahconnoc.

Since this is an i2c device, you'll need to use the [i2c library](https://github.com/bromagosa/Snap4Arduino/files/1645768/i2c.zip) for Snap4Arduino.

The datasheet for the device can be found [here](https://cdn-shop.adafruit.com/datasheets/BST-BMP180-DS000-09.pdf), and the i2c interface specs start in page 19. In particular, page 21 has a table with the register address and the control register values for different readings.

Untested because I don't have one of these sensors at hand, but here's how you'd supposedly read the temperature out of this device:

[image]

i2c can be tricky, I hope it works for you! :)

jguille2, 2018-01-19:
Yes...
And I also point you to the [Arduino library](https://github.com/adafruit/Adafruit_BMP085_Unified/blob/master/Adafruit_BMP085_U.cpp) to help you in the calculations.

In fact, you can follow the library algorithm and leave aside the spec. This would be my way... but you'll see it takes a bit of time!

If you don't need immediately, I guess soon I'll do this work, because I'm waiting to doc some other keyestudio sensors. But I want to have these sensors before to test my proposals.

Joan

seteahconnoc, 2018-01-19:
Thank you! Hopefully, this does not exceed my skills. I 'll see.
@jguille2 Please, keep me in touch, if you 'll achieve something.

closed 2023-05-25 by jguille2