bromagosa/Snap4Arduino #222: cant download 'Wifi to Serial Sketch'

xia01eng, 2018-01-14:
Sadly I cant download 'Wifi to Serial Sketch' from [https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Wifi](url), which is in the field of 'Pushing the "Wifi to Serial" Sketch'. Could anyone tell me where I can get this?

bromagosa, 2018-01-16:
You're right, since snap4arduino.org doesn't exist anymore, the file was gone. Sorry about that!

I've pushed the file to this repo and changed the URL in the wiki page to point to it.

Thanks for reporting! :)

closed 2018-01-16 by bromagosa

xia01eng, 2018-01-23:
You are welcome. This project is really amazing. Just it still has some problems to be solved in the #224 issue which I asked yesterday. It looks like I havent connect the wire correctly. Could you give me some advice？