bromagosa/Snap4Arduino #225: Need to whitelist our copy of Snap4Arduino

ToonTalk, 2018-01-27:
This is similar to #128 but our reasons for wanting to host a copy are:

1. We need to use HTTPS and the security warnings discussed in #177 haven't been resolved. (And have caused confusion to our users.)
2.  We have a tool launching portal that ideally should just open ./snap4arduino/ so that it works the same whether one is running on localhost or github.io
3. We will be adding logging code to collect data for learning analytics.

Please add https://ecraft2learn.github.io/* or https://ecraft2learn.github.io/uui/* to Snap4Arduino/src/platforms/web/chromium/crx/manifest.json

jguille2, 2018-01-27:
Hi @ToonTalk,

I need to talk with @bromagosa about this... but it is not the best moment... because many "servers" questions are on the table.

We have an issue (an important issue for us) accessing to the cloud. You can see it, even in your iframed ecraft2learn snap4arduino. We have access to the cloud (logining, working with our own cloud files...) but the option "Open from URL" (to work with shared projects) is not running because a Miosoft issue (also, another whitelist).

Changes into SnapCloud are comming soon. And then, it will be the time to define wich Snap4Arduino (and also Snap) services/sites are supported by the team. And this will define the whitelists (for snapCloud, plugin...)

I think you have three choices:

1. You know you can make your testing plugin. If you edit locally _manifest.json_ adding your domains, you can _build_ the chromium platform version. You will not get a crx file... but a "plugin folder" fully functional.
  To install a "folder extension" in Chrome,  users have to mark "dev mode" in `chrome://extensions`, and they can drag an drop folder extensions.
  Yes, I know this is not a definitive solution for you.

2. If you need to hack Snap4Arduino and build your own site, I think the right solution is to build your own plugin (an "ecraft2learn plugin"). You can take full advantage of the Snap4Arduino code (and the plugin creation) and adapt it to your needs.
  We can help you if there was any problem... But it would seem more consistent: a ecraft2learn plugin to play with the ecraft2learn service...

3. And the other choice is to define ecraft2learn as an official Snap4Arduino site. If you don't need many 'hacks' (and also we are opened to think about adding  features to the main trunk), i think it is the best solution.
  But then, we must talk about this... and it's better to wait to close cloud issues.

Anyway, I write down the _SSL certificates_ tip to my list. You are right our official online service needs to avoid those security warnings.

Continue...

Joan

ToonTalk, 2018-01-28:
Thanks for the response. I'll bring up the options you listed with the rest of the technical team.

bromagosa, 2018-02-13:
This is fixed in the dev version :)

closed 2018-02-13 by bromagosa

ToonTalk, 2018-02-14:
Thanks for the update.

github.io works fine with HTTPS - our problem is the plugin doesn't work. We are looking into creating our own (solution 2 in your post of 18 days ago).

jguille2, 2018-02-14:
Ok,

I leave the ticket open till I take the SSL warnings issue...

And maybe, after the cloud migration (now we are overwhelmed) we'll talk about whitelist and collaboration between projects.

Joan

ToonTalk, 2018-03-10:
Until the SSL warning is resolved we've decided to use your suggestion #1: install a "folder extension" in Chrome. Thanks again for your help.