bromagosa/Snap4Arduino #227: Firmata communication paused when Snap4Arduino's window is minimized

jpliew, 2018-02-05:
When running a simple blink 

[image]

I can see that the communication is working TX and RX LEDs blinking and LED on D13 also blinking. As soon as the main window of Snap4Arduino is minimized, the communication paused (NOT terminated), and LED stopped blinking. When window size is un-minimized, communication start again and work as normal.

Windows 8.1
Snap4Arduino 1.2.4

Anyone has this issue?

Cheers

jguille2, 2018-02-05:
Hi @jpliew ,

Yes, and this is not a Windows issue, nor an Snap4Arduino issue... I'm testing and Snap! itself has this behavior: it pauses execution when it is minimized.

I had not noticed this. I want to think better to see if that makes sense or it needs to be amended. I close this ticket and I [open another one](https://github.com/jmoenig/Snap--Build-Your-Own-Blocks/issues/2006) in Snap!

Thanks for your contributions!

Joan

closed 2018-02-05 by jguille2

jguille2, 2018-02-15:
Hi @jpliew!

Only note that this will be fixed in the next version (1.2.5, next Friday) Thanks Bernat!

As we commented in [the Snap! issue](https://github.com/jmoenig/Snap--Build-Your-Own-Blocks/issues/2006), _--disable-raf-throttling_ flag is implemented. This background execution is certainly better ... and we already have it in our desktop version.

commit 8de6e5c5 (2018-02-14): fixes #227. It was closed but never fixed