bromagosa/Snap4Arduino #229: Is there a way to use a RFID device such as "MF-RC522"?

seteahconnoc, 2018-02-16:
Is there a way to use a RFID device such as "MF-RC522"?

I am just aiming to get in touch with this kind of technology. A simple reading of the individual ID would be sufficient. Any suggestions are welcome.

seteahconnoc, 2018-03-03:
I think, as a first approach to this topic, I am going to try it this way ...

https://www.heise.de/developer/artikel/Arduino-spricht-RFID-3464780.html

... to get a tag's ID, and this way ...

http://www.instructables.com/id/How-to-log-data-from-the-command-line-with-Arduino/

... to get the data from the arduino to my computer. Afterwards, I could use `Snap!`'s "http block" to import the data to my project. As said, it 's a first approach.

Any further suggestions would be great.

jguille2, 2018-03-04:
Hi @seteahconnoc,

I see MF-RC522 can use SPI or I2C protocols.

Then, you can use it wit StandardFirmata in Snap4Arduino using I2C connectors. Now (in current 1.2.5 version) you'll find the blocks into i2c library.

But you can not use (without any customization) spi protocol (that is used in your linked example).

So, you must search for i2c address and commands. I'll try to report more info... but this is the starting point.

Joan

seteahconnoc, 2018-03-04:
@jguille2 As always: Thank you for your quick and helpful response.

jguille2, 2018-03-04:
Hi @seteahconnoc, you are welcome :smile: 

But unfortunately, maybe i2c is not a way for you. As I said, RC522 xip support I2C protocol (also SPI and Uart), but I've seen this configuration is often done via hardware. You can find modules like [PN532](https://dangerousthings.com/wp-content/uploads/PN532_Manual_V3-1.pdf) with a config switch... but looking at MRFC522... it is usually  SPI configured by hardware.

You can find [information](https://forum.arduino.cc/index.php?topic=442750.0) to change this hardware setting but maybe this is too complicated (modifying the board).

There will not be an early SPI implementation in Snap4Arduino (firmata.js is not supporting it). So you must change to an I2c model or make the hardware modification.

Joan

closed 2018-03-04 by jguille2

seteahconnoc, 2018-03-05:
Changing the device is no problem for me. I 'm just getting to know this kind of technology. So, buying a PN532 device would be OK. At first, I would try it this way via I2C:

http://wiki.sunfounder.cc/index.php?title=PN532_NFC_RFID_Module

You wrote "you must search for i2c address and commands" (to get things running via Snap4Arduino?). Since, I 'm just getting to know the technology, this would be challenging, but I 'll give it a try.

Thank you, so far.

jguille2, 2018-03-05:
Perfect!
I would like to help you with more details... but it is difficult without a device to test it.

Anyway, you could test the i2c device with our library
[image]

And you have two sources to get info and test...

- The [MFRC522 spec](https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf) in 8.1.4 (page 16)

- Or looking at [Adafruit PN532 lib](https://github.com/adafruit/Adafruit-PN532). Then, you can see the commands in [Adafruit_PN532.ccp](https://github.com/adafruit/Adafruit-PN532/blob/master/Adafruit_PN532.cpp#L1708) and constant values in [Adafruit_PN532.h](https://github.com/adafruit/Adafruit-PN532/blob/master/Adafruit_PN532.h#L39).

Continue...

seteahconnoc, 2018-03-06:
This sounds interesting. Thanks.

seteahconnoc, 2018-04-29:
@jguille2 It took a while, but now, I found the time to get in touch with a PN532 device. I successfully tested the I2C functionality via:

http://wiki.sunfounder.cc/index.php?title=PN532_NFC_RFID_Module

Right now, I would be thankful, if someone could give me a hand for using the device with Snap4Arduino. I took a look at the sources, you referenced. Sadly, they were not helpful for starting a first "hello world". Basically, I am looking for a first example for the usage of the blocks from the I2C library. I would be quite happy, if the RFID device could just identify a card's UID value. In the example file for the Arduino IDE, I had to deal with these lines:

`success = nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);`

and

`Serial.print("  UID Value: ");`
`nfc.PrintHex(uid, uidLength);`

seteahconnoc, 2018-04-29:
BTW, I guess, this issue should be re-opened.