bromagosa/Snap4Arduino #232: Older version of a project in the new cloud

DavidGasku, 2018-02-19:
Hi, I'm not 100% sure, but I think the latest migration messed up my account. This project is not in its latest version, and we would like to check if that's the case, and if so, try to get the latest version from some backup.

http://snap4arduino.rocks/run#present:Username=davidgasku&ProjectName=dibujafunciones)

Thanks.

bromagosa, 2018-02-19:
Hi! Unless you were working on this project between 5:00am and 5:30am on Saturday, I don't think it's possible that you're getting an older version. And even then it would be really unlikely.

Are you sure you didn't accidentally save it to local storage while the cloud was not available? You can check by looking in the Browser tab of the Open Project menu.

jguille2, 2018-03-13:
Hi @DavidGasku ,
Is this problem still there?

I have not notices about pending cloud issues.

DavidGasku, 2018-03-14:
Well, we lost that version, but we are slowly coding it back. I'm sure it was our mistake, but since it happened around that weekend, we thought it could be related.
Thanks anyway!

closed 2018-03-14 by DavidGasku