bromagosa/Snap4Arduino #233: Mac OSX version still 1.2.4

aplazaris, 2018-02-20:
I am downloading the new version - the zip file's name states "1.2.5", but the Snap4Arduino still shows 1.2.4!

bromagosa, 2018-02-20:
Hmm... does it say 1.2.4 in the "About Snap4Arduino" dialog or just in the title bar?

[image]

aplazaris, 2018-02-20:
[image]
Weird! Have a look at the screenshot attached.

bromagosa, 2018-02-20:
I see. Don't worry, we'll take a look and fix it. Your version is 1.2.5, but I forgot to set the version number in the package, that's all :)

bromagosa, 2018-02-20:
I've uploaded a patched version. Thanks for noticing and reporting! :)

closed 2018-02-20 by bromagosa