bromagosa/Snap4Arduino #234: I cannot apply my language file to Snap4Arduino

ljb7977, 2018-02-25:
I made a Korean language file (lang-ko.js) but when I build it, I cannot see that applied.
How can I fix this? I want to contribute Korean translation to Snap4Arduino project.

Here is the commit: https://github.com/ljb7977/Snap4Arduino/commit/a1699897cffd344ae05b0dfc2859868446a60a26

jguille2, 2018-02-25:
Hi Lee Ju Been,

Thanks for your translation. We will be happy to have Korean translation in Snap4Arduino.

Details...

- The main problem is a typo error.: missing the _end comma_ in line 169. You must change `'네트워크 시리얼 포트'` for `'네트워크 시리얼 포트',`. And then, you get your translation strings.
- Another problem  is with the _slot strings_. You must leave a space after them (separating words). For example, in line 94, you must change `'아날로그 입력 %analogPin번 핀',` for `'아날로그 입력 %analogPin 번 핀',` 

Then, you'll get this:

[image]

- And two minor details:
  - Take care of the indentation. A regular indentation (4 for original string and 8 for translation string) would be great.
  - You are working in your fork in the _master_ branch. It would be perfect to work in a _custom branch_ to manage the Pull Request.

Again, thanks for your contribution. We expect your next PR with these changes... or, if you prefer, just tell me, and I'llupload the changes (I already have them implemented locally).

Joan

ljb7977, 2018-02-26:
Thank you for the answer! I'll apply the changes and send a pull request :)

jguille2, 2018-02-26:
Hi Lee Ju Been,

Your pull request is merge. Full Korean translation for next release (or current dev versions)!!!

Thanks!

Joan

closed 2018-02-26 by jguille2