bromagosa/Snap4Arduino #236: If a USB-serial connection fails, Snap4Arduino needs to be restarted to retry

jan-gerard, 2018-03-08:
If my serial connection fails (I selected COM1 iso. COM4 while the Arduino was on 4) or the connection dropped somehow, I always need to restart Snap4Arduino to retry. Just retry clicking "Connect Arduino" does not give a response, even when I first click "Disconnect Arduino". It does not matter what program I made, it even occurs with an empty new project.
In an older version (1 year ago), reconnecting worked without restarting; I just needed disconnect/connect. Could this be solved? Restarting is rather inconvenient, as I have to save my project and reload it. It's cumbersome with a primary school class with 15 systems, where there is always one that fails to work, and another, and... etc.
I use a Arduino Nano + CH341SER USB-serial port driver + Snap4Arduino_desktop-win-32_1.2.5.zip . Which additional information do you require to tackle this?

jguille2, 2018-03-08:
Hi @jan-gerad ,

You are right there are some problems with some failed connections. I'll try to study this ... but there always seems to be some strange case (the failed connection remains "stuck"). When first connection is ok, disconnect/connect functions are also running ok.

In the meantime... some workarounds...
- The problem with the "stuck" connection is only in this sprite. You can create a new sprite and make the connection. This is fine in the beginning of the projectes... or when the sprite connectable has few blocks.
- You can force the connections by code. You can build a block for your students...
[image]

Keep open this issue... and I'll come back to this.

Joan

jan-gerard, 2018-03-11:
Thank! This helps as a workaround for this moment.

jguille2, 2018-03-13:
Hi @jan-gerard 
This issue is just fixed with #239. So dev version is already fine... and next release will not need any workaround.

So I close this issue. I thought I would also review _online_version (chromium) before closing ... but there are other related issues (about closing messages)... so, better leave it for another issue.

closed 2018-03-13 by jguille2