bromagosa/Snap4Arduino #241: HC-06 Bluetooth connection issues

segertonGH, 2018-04-18:
Hello,

I'm having trouble connecting an Arduino Nano to Snap4Arduino using an HC-06 module.  

I've put a voltage divider across RX to get the correct voltage.  And have the the module connected to pins 0 and 1 in the correct sense.  I've configured the HC-06 to BAUD-7 and also disabled the HC-06 connection echos.

I've written simple test script to send and receive data as fast as possible over the set up at 57600 baud.  The test script works every time and the connection is stable.

However, I can't get a connection with Snap4Arduino.  And I can't figure out why.  

I've tried pressing the reset button.  That doesn't help.

I've tried the Firmata Test program and when I try to connect the initial query gets through but then freezes, I'm left with a frozen application an information bar showing "COM4: firmata-2.5 Tx:73 Rx:219" 

I'm using windows 10, 64-bit. With Snap4Arduino version 1.2.5

I've also tried with an UNO, no difference.

Thanks in advance.

segertonGH, 2018-04-19:
Update.  I replaced the Bluetooth module with an FTDI module and have reproduced the same conditions as per above.  I need to press reset so S4A / Firmata test script can pick up the firmata firmware hello message.  Powering board independently, connecting only the TX/RX lines from the FTDI.  My test script is working fine, as before.  S4A / Firmata test script not connecting, and freeze after the firmware hello message.  However, once I connect the FTDI ground pin to the ground of the board everything starts working.  A clue perhaps? Let's see.

colingarbutt, 2018-05-07:
I've just been experimenting with an HC-05 module and Arduino uno and I've found that the Chrome extension can lock the com port under certain conditions that you can clear by stopping and starting the extension.
Colin

segertonGH, 2018-05-09:
Hi Colin, 

Thanks for the reply.  I should have mentioned that I'm using the standalone application.

I've continued to investigate and was able to monitor the serial transmissions to and from the Snap Application and my Arduino Board and have found the source of the problem.

When Snap4Arduino initiates a connection over serial it sends out a REPORT_FIRMWARE message and waits for the Arduino board to respond with a printFirmwareVersion() message to confirm the connection.  While S4A waits for a response it periodically broadcasts a REPORT_VERSION message.  

For some reason the Bluetooth connection misses the initial REPORT_FIRMWARE message and so does not respond with expected printFirmwareVersion().  It does see the REPORT_VERSION messages and replies correctly.

The solution is to press the reset button (as was already suggested).  Pressing the reset button resets the Arduino (it doesn't reset the Bluetooth Module, so the serial connection between S4A and the module is still active) and upon initialisation Firmata sends out a printFirmwareVersion(), which is then picked up by the waiting S4A.  

QUESTION for the S4A development team....

Can you resend the REPORT_FIRMWARE message if no replay has been received after say two REPORT_VERSION messages have been sent? This would seem a simple fix.

All the best.

adryannn, 2018-06-10:
Hi,

I'm not sure if we are using same vesion, but I have no problem to connect, using the below steps:

1) Load the below code on your arduino, so you can setup your hc-06 device:

char NOMBRE[10] = "HC-06";
char BPS        = '7';
char PASS[10]   = "1234";

void setup() {
  // put your setup code here, to run once:

Serial.begin(9600);
pinMode(13,OUTPUT);
digitalWrite(13,HIGH);
delay(10000);
digitalWrite(13,LOW);

Serial.print("AT");
delay(1000);

Serial.print("AT+NAME");
Serial.print(NOMBRE);
delay(1000);

Serial.print("AT+BAUD");
Serial.print(BPS);
delay(1000);
}

void loop() {
  // put your main code here, to run repeatedly:
digitalWrite(13, !digitalRead(13));
delay(500);
}

2)
>  after code is uploaded, make a connection between your machine and HC-06 device, using 1234 password.
> check your HC-06  device COM port, as you will need it when you will connect from application

3) open the Snap4Arduino application> Arduino tab> click on Connect Arduino and chose the COM port you have checked earlier.

Best regards,
Adrian

closed 2018-07-07 by segertonGH