bromagosa/Snap4Arduino #243: Can S4A resend the REPORT_FIRMWARE message...

segertonGH, 2018-05-09:
...if no printFirmwareVersion() reply has been received after a connection has been made? Say after two REPORT_VERSION messages?  This would seem to fix slow Bluetooth module connections that miss this initial REPORT_FIRMWARE.  This can be handled from the Firmata side on the Arduino, but looks messy.

All the best,

Simon.

bromagosa, 2018-06-01:
That sounds definitely feasible. I'm super busy with other projects at the moment but I'll try to find some time to look into it.

segertonGH, 2018-07-15:
The issue can be reproduced via a USB connection by installing a 10uf capacitor between the reset line and ground.  This prevents the Arduino from resetting when S4A tries to connect (simulating the scenario for (some/most?) Bluetooth connections).

closed 2023-05-25 by jguille2