bromagosa/Snap4Arduino #244: https://snap-cloud.cs10.org invalid certificate

DavidGasku, 2018-05-15:


jguille2, 2018-05-15:
Yes, we are sorry... and trying to solve this issue.

Snap4Arduino 1.2.5 is pointing to _snap-cloud.cs10.org_... and this is right... but right now we have problems with the server and the domain certification.

We've hacked _online version_ to avoid this problem. It's also a good workaround to get the projects stored into the cloud. In desktop version, you can also change the _cloud url_ (click right mouse button with _shift key_ pressed a you will see the _URL..._ option) but this change is not stored (you must repeat it in the program start).

We are trying to fix this problem... If this lasts, we will release a new version with the fix.

Sorry!

Joan

jguille2, 2018-05-16:
You can upgrade to Snap4Arduino 1.2.6 (released today) and the Cloud issue is gone.
Thanks for your patience!

Joan

closed 2018-05-16 by jguille2