bromagosa/Snap4Arduino #255: analog reading inverse result (in 1.2.6 not in 1.2.5) !!!

aplazaris, 2018-06-28:
Simple circuit, with photoresistor and blink, if analog reading is below value e.g. 300 (i.e. when it is quite dark), else LED is off. In the newest version, 1.2.6 the result is exactly the opposite: Light -> the LED is blinking, dark-> the LED is off. The correct is happening when using the previous version (just tried it) 1.2.5.
[image]
And, of course the output is correct when using the code in the Arduino IDE:
[image]

jguille2, 2018-06-28:
Hi @aplazaris ,
I can not reproduce your issue. I've tested with 1.2.6 and with current dev version. The behavior is what you expected.

What is your Snap4Arduino flavour (desktop linux, online chromium...)?

I understand your test is right (rejecting an electronic issue with the IDE test)... but can you check it again? Maybe something is changing between both tests?

Joan

aplazaris, 2018-06-29:
The circuit is this:
[image]
The weird thing is that (according to the scenario and the code mentioned before) in order for the LED to blink (in version 1.2.6) the operator should be "if analog reading **0>300**" for the scenario to work! Exactly the opposite of the correct statement (the reading should be less than e.g. 300, to indicate that it is getting darker). I have implemented the circuit and the scenario through Arduino IDE: works normally; Snap4Arduino 1.2.6: Needs the "greaten than" to work!; Uninstall 1.2.6 and install 1.2.5: Works ok!
Very very strange!

jguille2, 2018-06-29:
Which Snap4Arduino version (desktop linux 32 bits, online chormium,...)?

As I said, I can't reproduce the problem.

And you can test the problem again with more details? I mean...
- Is "set digital pin 13 to True" turning on the Led?
- And the L (onboard) Led?
- And what about "analog reading 0"? Which values are reporting? Dark values? Light values?

Continue...

aplazaris, 2018-06-29:
Version: Windows 64 bits (sorry, I forgot to mention). I will provide you some further details later, when I try this again.

aplazaris, 2018-06-29:
Now, that IS weird! I tried today with version 1.2.5. It worked normally, as yesterday.

Screenshots for Dark and Light (values are shown - upper right corner) follow:

[image]
[image]

Then, I updated (once again) to the 1.2.6, and all worked fine! I get the same values, and the expected LED behaviour - Off when analog reading values above 300; blinking when values below 300.

So, it seems ok. Sorry for the trouble.

jguille2, 2018-06-29:
No problem!

And let's continue enjoying with Snap4Arduino :+1: 

Joan

closed 2018-06-29 by jguille2