bromagosa/Snap4Arduino #256: Can't install it on Ubuntu 18.04 LTS

gentilmente, 2018-06-30:
Hi, on Lenovo v310.
Using Arduino IDE examples works.
When unzipping snap4Arduino and double clicking, nothing happend.
By executing in Terminal 
`rd2@rd2:~/apps$ '/home/rd2/apps/Snap4Arduino/Snap4Arduino' 
ln: fallo al crear el enlace simbólico '/home/rd2/apps/Snap4Arduino/libudev.so.0': El archivo ya existe
/home/rd2/apps/Snap4Arduino/nw: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory`
I tried the online version with extension installed in Chnromiun.
When clicking "connect Arduino" appears "select a port" but no port to select.

jguille2, 2018-07-01:
Hi!
First, let's got to your desktop version issue.

It seems a problem with Chrome libraries (Yes, desktop version uses nw, based in chrome). Could you check if you have installed _libgconf-2-4_ package? If this is the problem (you will find in Internet many reports about this, reinstalling chrome in different Ubuntu releases)... the solution is easy
`apt-get install libgconf-2-4`

I wait for your answer... and we can continue...

Joan

gentilmente, 2018-07-03:
That was easy!! thanks

closed 2018-07-03 by gentilmente

jguille2, 2018-07-04:
Hi @gentilmente,

And the online version? I think _libgconf_ is not affecting here. If you have problems, feel free to open another issue.