bromagosa/Snap4Arduino #257: Takes forever to launch with Windows 10

puttley, 2018-07-01:
I installed Snap4Arduino on my Windows 10 machine. It installs without any issues, but the app takes at least 2-3 minutes to launch when clicked. I've tried the 32 and 64 bit versions - both behave the same. If I wait long enough, snap does eventually open up. What could be causing this issue?

jguille2, 2018-07-01:
Hi,

One first question. Are you using last 1.2.6 version? I question this because some older version have a delay trying to connect to our old domain, that is not available.
I will try to test it tomorrow ... but maybe somebody else with a Windows available could report us their tests.

Joan

jguille2, 2018-07-05:
Hi @puttley ,

I can't reproduce your issue.

I am not a Windows user... but a friend confirm me 1.2.6 is running ok in his W10-64bits machine.

I wish I could do more test... 

  - Is the problem still alive?
  - Could you test it in different machines?
  - Could any other people report their own tests in W10?

Continue...
Joan

puttley, 2018-07-06:
Hi Joan,

Yes, I am still having the same issues. I've tried 32 and 64 bit versions. I will try a download to another W10 machine as you suggest. I'm also wondering if any others have experienced this issue and can provide some insight on the cause.

Much appreciated...

closed 2023-05-25 by jguille2