bromagosa/Snap4Arduino #261: Please release a new version that includes Basque translation before November

aldatsa, 2018-09-24:
First of all, I'd like to congratulate @bromagosa on becoming a dad! Felicitats!

I'm not in a hurry and I don't want to bother you in this special time of your life. I only want to let you know that the students of [Arizmendi Ikastola](http://www.arizmendi.eus/) are going to start using Snap4Arduino at the beginning of November and it would be great if a new release that includes the Basque translation could be available for then.

Thank you very much!

jguille2, 2018-09-24:
Hi Asier (@aldatsa )

Ok! We try to release a new versions in October. I need to check different details.

Anyway, if we have problems to do the release, I'll prepare a dev version (offline and online) to solve this problem and to be in time for you.

Joan

bromagosa, 2018-09-26:
Kaixo Asier, eta eskerrik asko! (was that correct at all?)

I concur with Joan. We'll definitely be able to publish a new version in time for you to use it in your classroom. October is when I'll get back to work, so between Joan and myself I'm sure we'll have time to work on this.

aldatsa, 2018-09-26:
That's great!
Zuri Bernat! (Literally "To you Bernat!", meaning "You're welcome Bernat!"). Yes, what you said is correct :)

jguille2, 2018-10-19:
Kaixo Asier! (@aldatsa)

Next 1.2.7 version (of course, with the Euskara translation) is planned for next Monday, October 22.

You can test it from [https://github.com/bromagosa/Snap4Arduino/releases/tag/1.2.7](https://github.com/bromagosa/Snap4Arduino/releases/tag/1.2.7)

Thanks!

Joan

closed 2018-10-19 by jguille2

aldatsa, 2018-10-19:
Great! Thank you very much Joan (@jguille2). Moltes gràcies!