bromagosa/Snap4Arduino #269: ChromeOS app can't login to cloud

gmitter, 2018-10-22:
Using Snap4Arduino for ChromeOS with a chromebook. Nothing related to cloud works. Signup, Login, etc. It seems the app can't connect to cloud.snap.berkeley.edu at all.

Additionally, If I save the project to "Browser", it is only available while the app is open. If I close the app and reopen it, the project is gone.

These features work fine with the online app.
Snap! 4.1.2.7

jguille2, 2018-10-22:
Hi @gmitter,
I keep this issue open because we have to take decisions about the chromeOs version of Snap4Arduino.

Anyway, I write some comments...
- The ChromeOs version had always some limitations because these chrome apps have no permission to do everything Snap! does.
- Last changes in Chorme apps policies have create more problems to build this distribution.
- Chromium version (not ChromeOs) is actually the best choice. If you can work online (with an Internet connection) this is certainly the workaround you need. As you said, cloud features are working fine, and all the Snap! elements too.
- To connect Arduinos with the Chormium version, our Chromium plugin is required (it is not the same Snap4Arduino extension for ChromeOS). More info in [our web](http://snap4arduino.rocks/#install)

We'll try to improve ChromeOS version soon... or maybe we will have to build from scratch another version to work offline with Chromebooks.

Thanks for your patience.

Joan

gmitter, 2018-10-22:
Thanks for the response @jguille2. 
As a test, I actually installed the chromium version manually in one chromebook and you are right, it worked better. 

The problem is, we are a school and although I tried many different ways of installing the chromium version across all students' chromebooks in my GSuite/ Google for Education platform, I just could not. Is there any kind of tutorial/ tips on how to get this working? Why isn't it @ the chrome web store?

jguille2, 2018-10-22:
Hi @gmitter,
You are right. It must be inside the chrome web store.

But, as I said, we had problems with the changes in GoogleChrome polices. It seems to block extensions like this, that can connect with the serial port.

We'll try to investigate this and improve some solution.

Thanks
Joan

bromagosa, 2018-10-22:
@gmitter it is at the webstore :)

https://chrome.google.com/webstore/detail/snap4arduino/ghipaghphhpfncbokoobcjlapbnceipg

We need to update to 1.2.7 though... will work on it this week.

gmitter, 2018-10-22:
Hi @bromagosa that's the one I'm using. This is not an extension, it's a ChromeOS app. 
Funny thing is: on the webstore, it says 1.2.6, but when I choose 'About...', it says 1.2.7 (see screenshot attached).  
Anyway, this is the app that doesn't connect to the cloud. If this is fixed, all my problems are solved! :)

[image]

jguille2, 2018-10-22:
Yes, we will try to answer soon to this problem (as I said).

Note that when you are using Snap4Arduino 1.2.6,  you are using Snap! 4.1.27. This is right.

Current version 1.2.7 is using Snap! 4.2.1.4.

Anyway, we'll try to fix this cloud issue and publish a 1.2.7 version to the Store.

Joan

gmitter, 2018-10-22:
Great, tks!

jguille2, 2018-10-22:
Hi @gmitter 

I've made some changes... and I've just uploaded _ChromeOs_ version to [1.2.7 release](https://github.com/bromagosa/Snap4Arduino/releases/tag/1.2.7). You will see _Snap4Arduino_web-chromeos_1.2.7_ file, with the _crx_ app.

Please, test this, and confirm that cloud is working for you. I've already tested this. Of course, you must drag and drop the crx file into the _chrome extensions_ 

Bernat will update this app into the Chrome Web Store and then, this issue will be closeable.

Thanks

Joan

bromagosa, 2018-10-23:
Hi @gmitter,

I've just pushed 1.2.7 to the web store. Please let us know whether it works well :)

Thanks!
Bernat.

gmitter, 2018-10-23:
thanks guys, I'll test it later on today and let you know how it went.

gmitter, 2018-10-23:
Hi I juste tested version 1.2.7 and unfortunately the problem persists..

jguille2, 2018-10-23:
Hi @gmitter,
I'm sorry.
I guess you have tested the new ChromeWebStore version... and the you still have problems with cloud features. We must fix this... but if you need the chromeOs versions, you can (in the meantime) use the [version posted in our web](https://github.com/bromagosa/Snap4Arduino/releases/download/1.2.7/Snap4Arduino_web-chromeos_1.2.7.zip) (this is just the chromeos app, not the chromium plugin... and I tested into a Chromium browser and it runs ok).

Give us some time... and we will post here the news about the WebStore version.

Joan

gmitter, 2018-10-23:
yes @jguille2 I'm using the Chrome web store version. The problem is in my environment I need a chrome store version to be able to distribute the app to all my students, the 'manual' installation won't work for me..

jguille2, 2018-10-24:
Hi @gmitter,

This issue has been fixed and now, current chromeOs version (also the version published on the ChromeWebStore) is connecting to the Snap! Cloud.
Can you test it?
Thanks!

Joan

gmitter, 2018-10-24:
Guys, this is what I call an awesome support! Thank you very much, everything working perfectly. You guys rock!

closed 2018-10-24 by gmitter