bromagosa/Snap4Arduino #272: Distance sensor not working in Chrome

adryannn, 2018-12-01:
Hi team,

I have observed that using the Chrome extension (last version 1.2.7) the distance sensor does not showing the value. Using the same setup (firmata code/hardware/block) is working fine in the desktop app version (1.2.7).

I have started to work again on and old project, and I think firmata is at least 3 years old. Do you think this may be the issue? I thinking to this because,I have also encounter a lots of issue with BT connectivity (HC-06 with arduino uno), either with desktop application or Chrome version.
The only workaround for Chrome extension to connect with success in 90% of the cases was to start using following steps:
1) Open Chrome Extension
2) Refresh the Snap4Arduino connector
3) Connect to the available port 

If using desktop application, I need try several time to connect or restart application. I have absolute no problem if using and old version (from 2015).

I don't want to mix the topics, if you believe it make sense to open a new topic for BT connection, I will do that. I put it here because I believe the firmata version could be the reason for both.

Thanks,
Adrian

[image]

adryannn, 2018-12-03:
Hi Joan,

Thanks for the quick feedback.

>>I'm using Chromium plugin with Snap4Arduino online, not ChromeOs app. I have same hardware and firmata/code, and only in the online version the Distance sensor is not working.

inside firmata code is very simple:
Ultrasonic ultrasonic(13,12); // (Trig PIN,Echo PIN)  //for Ultrasonic - define pins
...
 case 0x09: //For Ultrasonic - Get the distance
    Serial.write(START_SYSEX);
    Serial.write(STRING_DATA);
    pinMode(13,OUTPUT);
    pinMode(12,INPUT);
    Serial.println(ultrasonic.Ranging(CM)); //ultrasonic.Ranging(CM)
    Serial.write(END_SYSEX);
    break;

the screenshot with Snap block is already uploaded in the previous comment.

>>Regarding BT, just to keep same laptop/OS, I have downloaded latest version and test it again:
Same hardware (HC-06 with arduino nano)
Lenovo t61 - Windows 7 32bit

old version 1.0.7 beta - I can connect without any issue
new version 1.2.7 - I can connect only if I press reset button, while is Snap4arduino is connecting to bluetooth port...

Anyway, I will try with the SA5firmata and let you know the results.

PS: I'm using other software to program arduino, which can be installed on mobile, and I have no problem with BT connection, but I hope the connection will become stable in Snap4arduino as I think this has more potential as is highly customizable  and I can add my own blocks.

Thanks,
Adrian

jguille2, 2018-12-03:
Hi Adrian,
Thanks. Your feedback with SA5Firmata will be appreciated.

That 1.0.7 beta version is too old! And yes, different changes over firmata... Then, two things to consider:

- I see the web version (with the chromium plugin) does not accept extra "sysex" definitions. Then, you must use SA5 blocks if you want to work with this web version.
- And for the desktop use and the BT problems, you can test to change your firmware. I guess your code (the sysex definition) is written inside and old firmata. You can use a fresh StandardFirmata and add the code inside. 

Anyway, I hope SA5 fix both issues.

Joan

closed 2023-05-25 by jguille2