bromagosa/Snap4Arduino #275: Problema con chromebook

moisesmf, 2019-01-10:
Hola, no consigo hacer funcionar snap4arduino_web-chromium-plugin_1.2.7.zip en un chromebook, con el archivo snap4arduino_web-chromium-plugin_1.2.6.crx no tuve ningún problema. Puede ser que sea por el formato? Me podrías facilitar el plugin 1.2.7 en crx?

Gracias!!

jguille2, 2019-01-10:
Hola!
El problema con el crx es que el cambio de políticas en las versiones de Chrome/chromium daba problemas.

Mira de seguir las instrucciones:
-Descomprimir el archivo
-Abrir la página de extensiones
-Seleccionar el "modo desarrolador"
-Arrastra la carpeta descomprimida a la página de extensiones.

Si no te cuadra (no lo estoy probando en un chromebook, sino en un navegador chromium), insiste y miraremos de buscar una solución.

Joan

moisesmf, 2019-01-15:
Hola Joan, he probado varias veces las instrucciones y no hace nada. 

El Chromebook está administrado por un dominio, igual es por eso. 

No encuentro la forma de hacerlo funcionar, con el archivo .crx lo añadía a la consola de administración y no tenía ningún problema.

Muchas gracias!!

jguille2, 2019-01-16:
Hola!
Hemos podido conseguir un Chromebook para hacer pruebas y hemos podido comprobar que el plugin funciona correctamente. Comento detalles:

- Como ya comentamos, ahora ya no deja subir la extensión vía crx. Es por eso que en la última versión hemos publicado un zip con la carpeta (y no el crx).
- Tanto en un Chromebook como en un navegador Chrome/Chromium hay que:
    - Ir a "chrome://extensions"
    - Activar el "modo desarrolador"
    - Arrastrar la carpeta que tiene el zip sobre esta página
    - Y comprovar que el plugin estaba  instalado correctamente.
- Con el Chromebook nos ha costado poder "arrastrar la carpeta" pero supongo que era porque no somos usuarios de estos dispositivos. Concretamente nos ha costado descomprimir el zip y tener la carpeta (es muy fácil abrir el zip y acceder al contenido y los distintos archivos... pero no tan fácil obtener la carpeta descomprimida). Lo hemos hecho via "Drive" pero supongo que hay otras maneras (tambien se puede descomprimir en otro dispositivo y copiar directamente la carpeta).

- Por último, comentar que si tienes aulas de Chromebooks quizá gestionas las actualizaciones conjuntamente. En ese caso, podrías realizar la instalación en un solo dispositivo y que todos los otros ya lo tengan en sincronizarse las actualizaciones.

Joan

moisesmf, 2019-01-17:
ok Joan, creo que el problema es que el chromebook está administrado por un dominio. Seguiré intentándolo. 

Gracias!!