bromagosa/Snap4Arduino #276: Problemas al abrir proyectos

profesorsallepto, 2019-02-01:
Buenas tengo varios proyectos en snap4arduino que no puedo abrir pues no puedo acceder a las carpetas accedo solo a la lista de proyectos realizados en ese ordenador. ¿Sabe alguien en que carpeta se guardan o como puedo acceder para añadir esos proyectos?

Muchas gracias

jguille2, 2019-02-01:
Hola @profesorsallepto ,

Entiendo que guardabas los proyectos en el "Navegador".

Has de abrir Snap4Arduino desde el mismo usuario (de la sesión del sistema operativo) que lo habías guardado. Si utilizas Snap! pasa lo mismo, pero tambíen has de abrirlo desde el mismo navegador.

Joan

profesorsallepto, 2019-02-07:
muchas gracias

-- 

*Si recibes este correo es porque eres alumno del profesor Luis
Fernández-Sanz Serrano de el Colegio La Salle Santa Natalia.** La
información incluida en este correo electrónico es confidencial y va
dirigida únicamente al destinatario del mismo. En el supuesto de que usted
no fuera el destinatario, le solicito que me lo indique y no comunique su
contenido a terceros, procediendo a su destrucción. Le informo que está
totalmente prohibida cualquier utilización, divulgación, distribución y/o
reproducción de esta comunicación sin autorización expresa en virtud de la
Legislación vigente. Gracias*

closed 2019-03-26 by jguille2