bromagosa/Snap4Arduino #277: I can't loaded standard firmata, s4a says:

SUperMattew043, 2019-02-12:
I try to loaded standard firmata on my arduino but there is an error, if i try to load firmta arduno says error during compilation. if I try to connect arduino whit s4a, on  connect arduino, COM4, s4a says: cuold not talk to arduino in port 4,che if firmata is loaded, I also try whit the other port but nothing...
what I can do?

jguille2, 2019-02-13:
Hi @SUperMattew043,
First of all. If you can't upload a Firmata firmware, don't try to connect the board to Snap4Arduino, because it has no sense.

You must check the "compilation error". If you are using the StandardFirmata (no hacked) the compilation will have to be ok. Maybe you haven't choose the right board...or maybe you are using a board not supported by firmata? Anyway, you can check the complete error message.

bromagosa, 2019-02-13:
Can you send us the complete error message from the Arduino IDE? Also, what board model are you using?

closed 2023-05-25 by jguille2