bromagosa/Snap4Arduino #278: ChromeOS 1.2.7 release missing Basque translation

aldatsa, 2019-03-20:
The ChromeOS 1.2.7 release of Snap4Arduino is packaged with Snap! 4.1.2.7 but it should use 4.2.1.4 as the rest of builds. This is a problem for us, as the Basque translation of Snap! was added in the version 4.2.1.
We would really appreciate if you could re-release the ChromeOS 1.2.7 version.
Thanks!

jguille2, 2019-03-21:
Oh yes! ChromeOS version is not well maintained.

I'll try to explain current situation.
- ChromeOS is a very limited version. The main problem is that ChromeOS apps block injected JS code, and so, our JS block is not running and also, a lot of libraries and Arduino extensions are not running.
- For these limitations, ChromeOS version hasn't been updated. It was dreprecated in favor of the "online version" and the Chromium plugin.
- In past releases it was not updated, but we keep it there because many people continued to use it.
- To make matters worse, Google has discontinued the plugins for its browser. And our preferred solution is now not clear.

Conclusions...
- Is a solution for you to use the Chromium plugin with the online version?
  - Note that if you need to work offline, you can also download it.
  - Also note that if you have problems with the plugin, you can use the "developer mode" inside the Chrome/Chromium extensions and load the plugin folder.

- If this solution is not right for you, we will try to rethink current ChromeOS version situation. Soon, we'll have Snap! 5 so, we will have to upgrade Snap4Arduino.

I wait your comments about the Chromium plugin solution.

Joan

jguille2, 2019-07-27:
Hi Asier (@aldatsa ) and Jaume,

As you know, ChromeOS version is not well maintained, because different limitations with the Javascript execution.

I encourage everybody to use "chromium" version, even inside ChromeBooks. Then the "chromium plugin" is only to allow board connections (using chrome/chromium browser)... but you can run ['online' version](http://snap4arduino.rocks/dev/#) from any browser... and you can play it offline because "chromium version"  also has all the "web files" (just "www" folder, where you can open "index.html" with your browser).

Anyway, we have updated our "chromeOS" version with all the changes in Snap4Arduino5 (and then, all new Snap!5 features). On September 2, release date, we'll see to update _Chrome web store_ but now, you can test our Release Candidate versions. [Here you are!](https://github.com/bromagosa/Snap4Arduino/releases/tag/5.0)

Only one note more... You will see "crx" and "folder" distributions. Use "crx" distro (just drag and drop that crx file to your chrome/chromium extensions). But if you have problems doing it, change to "developer mode" and use the "folder distro".

That's all!

Joan