bromagosa/Snap4Arduino #279: Can't connect to Arduino in /dev/ttyACM0

astoctas, 2019-03-25:
In debian 9, I can connect to /dev/ttyUSB0 (Arduino Nano) but not to /dev/ACM0 (Arduino Uno/Mega)
All have the same firmware uploaded.
Permissions are the same:

$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 mar 25 15:47 /dev/ttyUSB0
$ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 mar 25 15:42 /dev/ttyACM0

I am on dialout group.
I've tried Desktop and Online versions (with chrome plugin), having same results.

Any suggestions?

Regards!

jguille2, 2019-03-26:
Hi @astoctas 
I don't see any problem with the info reported. I ask you some questions...

-First question. Are you also uploading Firmata with the same debian9? Because if you can connect Arduino IDE with both boards, surely the problem is not with the permissions.

- When you try to make the connection, what is the error message? Because I guess Snap4Arduino is detecting the port, isn't it?

- And are your UNO/Mega 0 and 1 pins connected anywhere? Because this could be bothering the connection. Sometimes we have boards with a Bluetooth module connected to RX/TX to implement that serial connection by BT... but this can bother our main USB serial connection.

- And waiting for your answers... a last comment. If the problem is with an error message like "Could not talk to Arduino in port /dev/ttyACM0 Check if firmata is loaded.", try to do a Reset  (board reset button) during the connection attempt.

Continue,
Joan

astoctas, 2019-03-26:
Hi @jguille2, 

Thanks for the reply.  I will answer and add some info:
It isn't about permissions because I can upload firmware to these boards and can use them with other applications, also with firmata installed.
1. The error message is "Could not talk to Arduino in port /dev/ttyACM0 Check if firmata is loaded"
2. Both boards have no conections on any pins. 
3. I've already tried these resets as it is said in other posts.

Also I add that in Windows 10, these boards are working (COM7 in my case), so it isn't a firmata or board issue.  It seems to be a debian/linux or a s4a (linux version) issue.

Thanks in advance.
Regards!

astoctas, 2019-03-26:
Hi @jguille2,

I've found that it happens just after uploading firmware to the board. Even closing Arduino IDE.  USB port  keeps blocked somehow. (Works with other applications but not with s4a)
Restarting linux solves the problem.  I don't know if it is a debian/linux  or a s4a (linux version) issue, but keep in mind when upload Firmata to the board using /dev/ttyACM0.

Thank you very much.
Regards.

jguille2, 2019-03-26:
Oops!
@astoctas This is strange.
Note that all the Snap4Arduino developers are linux users... and we have always problems to test Windows versions... but Ubuntu/Debian is ultratested.

The only time I saw problems with a busy port was connecting over Bluetooth, because some distros starts a daemon (I think it was for internet wireless connections) that occupied those ports.

But this is not the case. Do you use the Arduino IDE or the online Arduino Create? Or even using Arduino IDE, has you other browser plugins (like those for Arduino Create) that use serial connections?
And another test. Without restarting linux, do you test to unplug and plug again the USB plug? If it is busy with other services, maybe it gets another port (ACM1, 2...) and allow the Snap4Arduino connection.

Continue...

Joan

astoctas, 2019-04-03:
Hi @jguille2, 

I'm afraid Atom + Platform IO was blocking USB port after upload firmware.  Even closing Atom.  By now the only way to unblock it is resetting PC.  
I will tell you if I have any news.

Thanks, regards!

jguille2, 2019-04-03:
Hi @astoctas,
This makes sense.

Let's leave this issue opened, and your own answers will be interesting for future issues. Thanks!

Joan

closed 2023-05-25 by jguille2